meshtastic/firmware #5826: [Bug]: ROUTER_LATE on LongFast - Packets sometimes dont TX

Talie5in, 2025-01-12:
### Category

Other

### Hardware

Rak4631, Linux Native

### Firmware Version

2.5.18.89ebafc

### Description

Seeing issues where a Packet wont TX in ROUTER_LATE under LongFast if a second or additional duplicate packet comes in and it pushed the packet wait out, we don't eventuate to a "[RadioIf] Started Tx"

If however, we don't see a duplicate packet, it will queue for TX.

[DoesTX.txt](https://github.com/user-attachments/files/18387933/DoesTX.txt)
[NeverTXd.txt](https://github.com/user-attachments/files/18387934/NeverTXd.txt)

Started discussing this with @erayd in Discord.

erayd, 2025-01-12:
I suspect this may be due to a full TX queue, and have found a bug in the drop logic. Fix for that part incoming...

erayd, 2025-01-12:
@GUVWAF What are your thoughts on increasing the size of the TX queue? Testing has shown some packets on `LONG_FAST` are sitting in the queue >10 seconds waiting for an opportunity to send, even just in the standard window (i.e. non-delayed packets).

Talie5in, 2025-01-12:
Other things to note is it gets to a point where, the TX queue is full, and you never see [RadioIf] attempt to ever send another TX packet, gets to a point where the test node behind the ROUTER_LATE see's no new nodeinfos or anything once it hits that point, cant send any messages out, or request position or do a traceroute - only way out is to restart.

```
WARN  | 15:55:46 2625 [Router] TX queue is full, and there is no lower-priority packet available to evict in favour of 0x6afb9dd5

WARN  | 15:55:53 2633 [Router] TX queue is full, and there is no lower-priority packet available to evict in favour of 0x580f37c4

WARN  | 15:53:01 2460 [Router] TX queue is full, and there is no lower-priority packet available to evict in favour of 0xe10e6e6c

```

GUVWAF, 2025-01-12:
@erayd Increasing the Tx queue size might be worth considering, but let's first try to narrow down the causes of the stalls and fix them. Dropping packets might be justified if the mesh is really busy.

erayd, 2025-01-12:
Sounds good.

On the stalls, I think the cause is identified. @Talie5in and myself are currently thrashing the latest in #5827 to make sure.

PR #5827 (2025-01-12): Bugfix for low-priority packet replacement when TX queue is full
Add the intended replacement packet (not the packet that was just removed, which was clearly a typo) to the TX queue after dropping a lower-priority packet. Fixes a use-after-free.

Also corrects the function comment, and adds some debug logging when packets are dropped due to a full TX queue.

Fixes #5826.
closed 2025-01-12 by thebentern