meshtastic/firmware #5829: [Feature Request]: CH341 USB device scanning (auto-enumeration)

vidplace7, 2025-01-12:
### Platform

Linux Native

### Description

Currently `meshtasticd` requires manual config file changes after installation, which can be daunting to users who are not experienced with Linux machines.
With the coming release of CH341 USB dongles, we have the opportunity to provide a more simplified out of box experience.

`meshtasticd` should probe the usb bus for devices that match known VID and PID values, and select the appropriate pinmap if a known value can be found.

This should become the default configuration of `meshtasticd`, while still allowing advanced users to change their configs for SPI Pi Hats / etc.

vidplace7, 2025-01-12:
@psiegl would you consider taking a look at this? :heart:

psiegl, 2025-01-12:
Well, what can I say: You are truly right. It took me quite some afternoons to gdb myself through the code, figuring out what's going on. And I see your concern, meaning it is though to set up the right config. E.g. https://github.com/meshtastic/firmware/blob/master/src/platform/portduino/PortduinoGlue.cpp#L532 . Imagine you are missing the `General:` part in your homebrew config. Suddenly, meshtastic doesn't have the default general config flags, and well it doesn't survive and won't tell you :) . Just one of the issues, I stumbled into. My first thought was to propose a general config, that Meshtastic always comes with, while the user can 'overlay' his/her. However, you are right: Ideally meshtastic tries at least its best to scan for known configs and solely relies on the user interaction in case he/she enforces it. I will for sure propose changes to the PortduinoGlue.cpp. Maybe I have a good idea, how to bring in your idea.

mverch67, 2025-01-13:
Can't this be done automatically using _udev rules_ (assuming with _coming release_ you mean those with a unique serial numbers) ? E.g. something like:

`SUBSYSTEM=="usb", ACTION=="add" ATTRS{idVendor}=="1a86", ATTRS{idProduct}=="5512", DRIVERS=="ch341", PROGRAM="/usr/sbin/meshtastic-macgen %n" RUN+="/usr/sbin/meshtasticd -h %c -d ${HOME}/.portduino/%c -c /etc/meshtasticd/config-%n.yaml" MODE:="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_PORT_IGNORE}="1"`

fifieldt, 2025-01-13:
I believe that's a good idea. The package could install a file in /etc/udev/rules.d/

garthvh, 2026-09-03:
Closing: CH341 USB device auto-detection was implemented in PR #6446 (src/platform/portduino/PortduinoGlue.cpp), which probes the USB bus and device tree for known hardware and auto-selects the config. This shipped in the 2.7.x line, predating 2.8, and remains in place in the 2.8 alpha (v2.8.0.47db0e3).

closed 2026-09-03 by garthvh