meshtastic/firmware #5839: [Bug]: Critical fault #12 (flash filesystem corruption) on Heltec T114 v2

dahanc, 2025-01-13:
### Category

Other

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.13.1a06f88

### Description

Back in November 2024, my T114 showed a Critical fault #&NoBreak;12. I rebooted it and it seemed to work OK, but a few days later, it got into a boot loop. The serial debug output was:
```
INFO  | ??:??:?? 6 Adding node to database with 1 nodes and 147308 bytes free!
DEBUG | ??:??:?? 6 Expanding short PSK #1
INFO  | ??:??:?? 6 Wanted region 1, using US
INFO  | ??:??:?? 6 Saving /prefs/db.proto
lfs warn:314: No more free space 224
ERROR | ??:??:?? 6 Error: can't encode protobuf io error
ERROR | ??:??:?? 6 LFS assert: head >= 2 && head <= lfs->cfg->block_count
ERROR | ??:??:?? 6 LFS assert: head >= 2 && head <= lfs->cfg->block_count
ERROR | ??:??:?? 6 LFS assert: block < lfs->cfg
```
Then about a week later, another of my T114s also got into a boot loop. This one doesn't have a screen, so I don't know if it also had a Critical fault #&NoBreak;12. I erased the flash, installed 2.5.13.1a06f88, connected the USB port to a PC, and logged all of the serial output, so in case it happened again, I could see what happened right before the first crash and reboot. After about a month, it rebooted:

```
INFO  | 20:29:31 2159141 [Router] Received nodeinfo from=0xa20a11c8, id=0x7d3b35e6, portnum=4, payloadlen=44
INFO  | 20:29:31 2159141 [Router] Node database full with 80 nodes and 109096 bytes free. Erasing oldest entry
INFO  | 20:29:31 2159141 [Router] Adding node to database with 80 nodes and 109096 bytes free!
DEBUG | 20:29:31 2159141 [Router] old user /, channel=0
DEBUG | 20:29:31 2159141 [Router] Update changed=1 user MKTX G2 Gateway/MKG3, channel=0
DEBUG | 20:29:31 2159141 [Router] State: ON
DEBUG | 20:29:31 2159141 [Router] Node status update: 22 online, 80 total
INFO  | 20:29:31 2159141 [Router] Save /prefs/db.proto
lfs debug:617: Bad block at 126
lfs debug:617: Bad block at 180
INFO  | 20:29:36 2159146 [Router] BLE Disconnected, reason = 0x8
DEBUG | 20:29:36 2159146 [Router] PhoneAPI::close()
lfs debug:640: Relocating 126 160 to 181 160
ERROR | 20:29:37 2159147 [Router] LFS assert: head >= 2 && head <= lfs->cfg->block_count
ERROR | 20:29:37 2159147 [Rou
```
And after the reboot, these is what it logged before rebooting again:
```
INFO  | ??:??:?? 3 Adding node to database with 1 nodes and 153800 bytes free!
DEBUG | ??:??:?? 4 Expand short PSK #1
INFO  | ??:??:?? 4 Wanted region 1, using US
DEBUG | ??:??:?? 4 Coerce telemetry to min of 30 minutes on defaults
INFO  | ??:??:?? 4 Save /prefs/db.proto
ERROR | ??:??:?? 4 LFS assert: block < lfs->cfg->block_cou
```

So it seems that what triggered it was the "Bad block" errors, and maybe the "relocation" code is buggy and corrupts the filesystem? In any case, the "Bad block" errors seem more relevant. From what I see in [lfs.c](https://github.com/geeksville/Adafruit_nRF52_Arduino/blob/master/libraries/Adafruit_LittleFS/src/littlefs/lfs.c), it looks like the "Bad block" message means a flash routine returned `LFS_ERR_CORRUPT`. However, I didn't see anything in [InternalFileSystem.cpp](https://github.com/geeksville/Adafruit_nRF52_Arduino/blob/master/libraries/InternalFileSytem/src/InternalFileSystem.cpp) that would return `LFS_ERR_CORRUPT`. So I think that means `lfs_cache_cmp()` returned `false` (e.g., [line 194 of lfs.c](https://github.com/geeksville/Adafruit_nRF52_Arduino/blob/master/libraries/Adafruit_LittleFS/src/littlefs/lfs.c#L194)).

I haven't looked into the details of how the caching works, but since I don't think the flash is going bad on either of my T114s (V2 hasn't been out that long), I wonder if something else in the firmware is corrupting the memory buffer used for the cache.

esev, 2025-01-13:
Do you use an Android phone?

dahanc, 2025-01-13:
Yes, I use an Android phone.

daniel-hofer, 2025-01-14:
Same Error on T-Echo. Logs attached. Use Android with 2.5.16 beta App.

[techolog.txt](https://github.com/user-attachments/files/18409539/techolog.txt)

ERROR | ??:??:?? 4 Error: can't encode protobuf io error
ERROR | ??:??:?? 4 Can't write prefs!
ERROR | ??:??:?? 4 Failed to save to disk, retrying
DEBUG | ??:??:?? 5 Opening /prefs/config.proto, fullAtomic=1
INFO  | ??:??:?? 5 Save /prefs/config.proto
DEBUG | ??:??:?? 5 Opening /prefs/channels.proto, fullAtomic=1
INFO  | ??:??:?? 5 Save /prefs/channels.proto
DEBUG | ??:??:?? 6 Opening /prefs/db.proto, fullAtomic=0
INFO  | ??:??:?? 6 Save /prefs/db.proto
ERROR | ??:??:?? 6 NOTE! Record critical error 12 at src/mesh/NodeDB.cpp:1220

Ztrby, 2025-01-14:
I think i have the same problem, my device is in a boot loop and i can't get it to start at all

esev, 2025-01-15:
From the initial report:

```
INFO  | 20:29:31 2159141 [Router] Save /prefs/db.proto
lfs debug:617: Bad block at 126
lfs debug:617: Bad block at 180
INFO  | 20:29:36 2159146 [Router] BLE Disconnected, reason = 0x8
DEBUG | 20:29:36 2159146 [Router] PhoneAPI::close()
lfs debug:640: Relocating 126 160 to 181 160
```

And from trying to reproduce the issue last night:

```
INFO  | 01:48:53 1293 [Router] Adding node to database with 80 nodes and 117812 bytes free!
DEBUG | 01:48:53 1293 [Router] Update changed=1 user North Fremont - Lakes&Birds Neigh/NFMT, id=0xe0f79ab8, channel=1
DEBUG | 01:48:53 1293 [Router] Node status update: 34 online, 80 total
lfs debug:617: Bad block at 94
lfs debug:617: Bad block at 178
INFO  | 01:48:55 1295 [Router] BLE Disconnected, reason = 0x8
DEBUG | 01:48:55 1295 [Router] PhoneAPI::close()
lfs debug:639: Relocating 94 93 to 179 93
INFO  | 01:48:56 1296 [Router] Save /prefs/db.proto
```

Both have something Bluetooth related happening around the same time as an update to the db.proto file.

The Bluetooth library also writes to the file system. https://github.com/geeksville/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/utility/bonding.cpp

Could there be some preemption happening that causes a LFS change to happen at the same time as another LFS change is happening?

esev, 2025-01-15:
I've been able to get some debug logs by modifying this code: https://github.com/geeksville/Adafruit_nRF52_Arduino/blob/4f591d0f71f75e5128fab9dc42ac72f1696cf89f/cores/nRF5/common_func.h#L154-L158

And adding `-DCFG_DEBUG=1 -DCFG_LOGGER=0`

```c++
#if __cplusplus
  extern "C" void logLegacy(const char *level, const char *fmt, ...);
  #define PRINTF(...)    logLegacy("DEBUG", __VA_ARGS__)
#else
  void logLegacy(const char *level, const char *fmt, ...);
  #define PRINTF(...)    logLegacy("DEBUG", __VA_ARGS__)
#endif
```

I'm not easily able to reproduce the issue though.

todd-herbert, 2025-01-15:
@esev It seems like you've got a lot more knowledge than me here, and it seems like there's some good clues in the thread so far. I've been testing around this issue today, and as far as I can tell: it occurs when Bluetooth disconnects unexpectedly during a flash write. 

The reason for the disconnection seems relevant.

A graceful disconnect, initiated by the phone, gives reason:
> `0x13` `BLE_HCI_REMOTE_USER_TERMINATED_CONNECTION`
If this type of disconnection occurs during a flash write, there is no issue.

Disconnection types that I've seen cause issues are:
>`0x08` `BLE_HCI_CONNECTION_TIMEOUT`
> `0x22` `BLE_HCI_STATUS_CODE_LMP_RESPONSE_TIMEOUT`

I see `0x08` when the device moves out of range.
I see `0x22` when the device is disconnected on `NRF52Bluetooth::shutdown`.
If either of these disconnections occur during a flash write, lfs reports bad blocks, and corrupts the flash data.

One workaround for the`0x22` seems to be to modify `NRF52Bluetooth::shutdown`, to poll for disconnection before continuing.
This doesn't get to the root of the issue, but does seems to reliably solve this one case.
```cpp
void NRF52Bluetooth::shutdown()
{
    // Shutdown bluetooth for minimum power draw
    LOG_INFO("Disable NRF52 bluetooth");
    uint8_t connection_num = Bluefruit.connected();
    if (connection_num) {
        for (uint8_t i = 0; i < connection_num; i++) {
            LOG_INFO("NRF52 bluetooth disconnecting handle %d", i);
            Bluefruit.disconnect(i);
        }

        // Wait for disconnection
        while(Bluefruit.connected())
            yield();
        LOG_INFO("All bluetooth connections ended");
    }
    Bluefruit.Advertising.stop();
}
```

I haven't found any workaround for `0x08` disconnections so far. It would be nice if there is a true solution to be found somewhere deeper down in Bluefruit / the softdevice, but that's probably beyond my skill level.

___

> I'm not easily able to reproduce the issue though.

It may depend on the phone used for testing, but here is how I have been able to reproduce the issue:
* Heltec T114 (V1)
* Android app 2.5.16
* No existing config required (erased before flash)

#### `0x22`
* Firmware: #5858
* Connect phone via Bluetooth
* Erase NodeDB

### `0x08`
* Modify #5858 firmware: add `nodeDB->saveToDisk()` to ButtonThread.cpp, so that user button pretty triggers flash write
* Pair phone via Bluetooth
* Place phone in unplugged microwave to shield RF (*sophisticated.. I know*)
* Node powered by laptop, for portable logging.
* Bring the node very near the closed microwave, allowing Bluetooth to automatically reconnect.
* Move away from the microwave. A short distance (~3m seems to be enough in my case).
* Before the Bluetooth connection is lost, press the user button, triggering the flash write. This seems to occur within several seconds, so it is fairly easy to time this without repeatedly pressing the button.

This is all a bit unorthodox, but using this method I seem to be able to repeatedly reproduce the logs from https://github.com/meshtastic/firmware/issues/5839#issuecomment-2591353588

thebentern, 2025-01-15:
> There are definitely some file accesses that happen when the phone connects.

Looks like we are likely going to need to make some modifications to Bluefruit's persistence. I would prefer it to pull more of those things into memory to cut down on file system IO if possible. That's a shoot from the hip answer though, until I dig in more.

> One workaround for the0x22 seems to be to modify NRF52Bluetooth::shutdown, to poll for disconnection before continuing.
This doesn't get to the root of the issue, but does seems to reliably solve this one case.

@todd-herbert I think that workaround is worth a PR even if it's not a comprehensive fix.

todd-herbert, 2025-01-15:
With the changes described in https://github.com/meshtastic/firmware/issues/5839#issuecomment-2591688048, I've grabbed some more detailed logs of what's going on in the `BLE_HCI_CONNECTION_TIMEOUT (0x08)` scenario (disconnect out of range). Not sure if there's anything to be made of it, but here's the info:
```
DEBUG | 12:14:45 87 [Power] Battery: usbPower=0, isCharging=0, batMv=4159, batPct=97
DEBUG | 12:14:47 89 [Button] Opening /prefs/config.proto, fullAtomic=1
INFO  | 12:14:47 89 [Button] Save /prefs/config.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_ERROR#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:617: Bad block at 30
[SOC   ] NRF_EVT_FLASH_OPERATION_ERROR#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:617: Bad block at 29
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0#
[GAP   ] Disconnect Reason: CONNECTION_TIMEOUT #
INFO  | 12:14:49 91 [Button] BLE Disconnected, reason = 0x8
DEBUG | 12:14:49 91 [Button] PhoneAPI::close()
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:639: Relocating 30 5 to 35 5
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:2526: Found desync 35 5
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
DEBUG | 12:14:49 92 [Button] Opening /prefs/module.proto, fullAtomic=1
INFO  | 12:14:49 92 [Button] Save /prefs/module.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
DEBUG | 12:14:50 92 [Button] Opening /prefs/channels.proto, fullAtomic=1
INFO  | 12:14:50 92 [Button] Save /prefs/channels.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
DEBUG | 12:14:50 93 [Button] Opening /prefs/db.proto, fullAtomic=0
INFO  | 12:14:50 93 [Button] Save /prefs/db.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
DEBUG | 12:14:51 93 [Button] State: ON
bool BLEAdvertising::_start(uint16_t, uint16_t): 377: verify failed, error = NRF_ERROR_INVALID_STATE#
DEBUG | 12:15:05 107 [Power] Battery: usbPower=0, isCharging=0, batMv=4148, batPct=97
```

esev, 2025-01-15:
That's very helpful @todd-herbert !

This may be odd timeout behavior in the hardware. I'll look into this more after work tonight.

- https://devzone.nordicsemi.com/f/nordic-q-a/87217/flash-operation-always-return-nrf_evt_flash_operation_error
- https://os.mbed.com/teams/Maintool/code/mbed-src-v4/raw-file/0e58554f11d5/targets/hal/TARGET_NORDIC/TARGET_NRF51822/Lib/s110_nrf51822_7_0_0/s110_nrf51822_7.0.0_releasenotes.pdf (page 6, Known issues)

The microwave idea and the save on button press should make debugging much easier.

todd-herbert, 2025-01-15:
Looking at the reference links you've collected, I think I might have a working solution, although possibly I'm just being naive.. I'm not sure what the full implications of it would be. It's late here now but I'll try to get a proof of concept pushed tonight so you can see the general idea and check if there's any merit in it.

todd-herbert, 2025-01-15:
Here's the changes that *appear* to fix the issue: [flash_nrf5x.c](https://github.com/geeksville/Adafruit_nRF52_Arduino/compare/master...todd-herbert:meshtastic-nrf52-arduino:reattempt-flash-ops)

```
DEBUG | 16:17:19 43 [Button] Opening /prefs/config.proto, fullAtomic=1
INFO  | 16:17:19 43 [Button] Save /prefs/config.proto
DEBUG | 16:17:20 44 [Button] Opening /prefs/module.proto, fullAtomic=1
INFO  | 16:17:20 44 [Button] Save /prefs/module.proto
DEBUG | 16:17:20 44 [Button] Opening /prefs/channels.proto, fullAtomic=1
INFO  | 16:17:20 44 [Button] Save /prefs/channels.proto
INFO  | 16:17:25 49 [Button] BLE Disconnected, reason = 0x8
DEBUG | 16:17:25 49 [Button] PhoneAPI::close()
DEBUG | 16:17:25 49 [Button] Opening /prefs/db.proto, fullAtomic=0
INFO  | 16:17:25 49 [Button] Save /prefs/db.proto
DEBUG | 16:17:25 49 [Button] State: ON
DEBUG | 16:17:25 49 [Power] Battery: usbPower=0, isCharging=0, batMv=4172, batPct=98
INFO  | 16:17:25 49 [DeviceTelemetry] Send: air_util_tx=0.002611, channel_utilization=0.156667, battery_level=98, voltage=4.172000, uptime=49
```

I seems to be able to trigger the problematic disconnect with no consequences. With the additional logging setup you described earlier I was able to add extra debug output to confirm that in the situation shown above the flash operations do fail, and are successfully reattempted.

> Edit4: Curious why this semaphore allows up to 10 concurrent accesses instead of just 1?

That is the big question.. I've replaced it with a binary semaphore with no immediately apparent consequences, but it's a bit over my head so I'm hoping someone more knowledgeable will know if there was some very good reason for that counting semaphore.

esev, 2025-01-15:
Yes! That is the exact same thing I was going to try. Nice work!

I'll dig-in more to the semaphore. I'm kind of wondering if everything behaves the same if the max value is decreased to 1. Maybe only one is actually permitted at a time and the changes serialize by waiting for `err != NRF_ERROR_BUSY`?

esev, 2025-01-16:
Finally got a chance to test. This is working perfectly on my device. I can trigger the issue and I see the retry happen and succeed. No more lfs messages!

Prior to testing your change, I also added some logging to show how many simultaneous callers passed through the original semaphore. It was only ever one at a time. That matches what I was expecting too, as I think the device can only perform one flash operation at a time reports NRF_ERROR_BUSY when one is in progress. I like your change to use a binary semaphore. It seems much less surprising to me.

esev, 2025-01-16:
One thing. I'm not 100% sure it is necessary to `assert` inside the flash operations. In theory LittleFS should detect the failure and relocate the failing blocks.

https://github.com/littlefs-project/littlefs/blob/v1.7.2/DESIGN.md

dahanc, 2025-01-16:
Very cool! I don't think I would have noticed that it was related to Bluetooth :)

esev, 2025-01-16:
For keeping track of debugging ideas for the future:

- I ended up adding a `switch (c)` statement inside this `if` statement: https://github.com/meshtastic/firmware/blob/f9876cfe9c52691b56305270e1246fdc6d72f37f/src/mesh/StreamAPI.cpp#L46-L47 I used the 'c' character to trigger `nodeDB->saveToDisk()`. That could be easily extended to trigger other events when debugging as well.
- A faraday bag (https://www.amazon.com/gp/product/B0BYM9VJMR) works well to trigger the Bluetooth timeout.

todd-herbert, 2025-01-16:
> Yes! That is the exact same thing (retries) I was going to try

That is reassuring to know that you had also independently reached the same conclusion about where we should go with this for a fix!

> Maybe only one is actually permitted at a time and the changes happen in series by waiting for err != NRF_ERROR_BUSY?

> Prior to testing your change, I also added some logging to show how many simultaneous callers passed through the original semaphore. It was only ever one at a time. That matches what I was expecting too, as I think the device can only perform one flash operation at a time and reports NRF_ERROR_BUSY when one is in progress.

Ahh that would make sense if `NRF_ERROR_BUSY` blocks it that way. For what it's worth, I didn't see any instances of `NRF_ERROR_BUSY` in the wild while testing the proto saving, but I imagine it's totally possible that the Meshtastic firmware just doesn't trigger that sort of a simultaneous write situation.

> One thing. I'm not sure it is necessary to assert inside the flash operations. In theory LittleFS should detect the failure and relocate the failing blocks. It might be okay to give it a chance to do that.

That's a very good point. I'd restructured a bit today to cut back on the amount of duplicate code, and in the process had gotten rid of the assert, out of a general fear of unintentionally disrupting some fs behavior. Hopefully there's nothing similar lurking in today's changes too. I've opened it up as draft at https://github.com/meshtastic/Adafruit_nRF52_Arduino/pull/1, and would very much appreciate your feedback if you spot anything we could do better.

tavdog, 2025-01-17:
I just got a another LFS_ASSERT crash.  Not boot looping this time though. I tried using https://github.com/todd-herbert/meshtastic-nrf52-arduino/tree/reattempt-flash-ops to see if it would bring it back up but it did not fix it. The only way I have been able to rescusitate it i to add FSCom.format() in to the arduino setup function. Then once it has been reformatted on first boot, re-flash with the format line removed.

todd-herbert, 2025-01-17:
When I was inducing that 0x22 ble disconnect failure on shutdown, I would see it eventually end up in that same situation, after I had induced seveal shutdown-corruptions, and the blocks had remapped several times. Maybe there's a clue there?

Did you give the device a full erase recently, or has it had a history of these corruption events already?

If it does lock up again, one idea might be to enable the extra logging as described by https://github.com/meshtastic/firmware/issues/5839#issuecomment-2591611038, to see if there's any interesting info. I imagine you could do this even after it locks up.

tavdog, 2025-01-17:
> When I was inducing that 0x22 ble disconnect failure on shutdown, I would see it eventually end up in that same situation, after I had induced seveal shutdown-corruptions, and the blocks had remapped several times. Maybe there's a clue there?
> 
> Did you give the device a full erase recently, or has it had a history of these corruption events already?
> 
> If it does lock up again, one idea might be to enable the extra logging as described by [#5839 (comment)](https://github.com/meshtastic/firmware/issues/5839#issuecomment-2591611038), to see if there's any interesting info. I imagine you could do this even after it locks up.

Yes my this device has had a history of this happening.  This is the third time I've had to deal with it.   First time it was reboot looping because I think the nodedb got too big but I also left it in the house and might have had a the BT distance thing happen.  My attempts at full erase didn't the fix the issue but I did try the full erase during the first and second occurences.

NireBryce, 2025-01-18:
Different device but adding more diagnostic data, this just happened to my Nano G2 a second time, clearing nodeDB fixed it after a few days of "nodeDB-full" warnings on the very populated Boston mesh. 

critical fault 12, but this time I caught it before the boot-loop.

early symptom is it forgetting the name I set.

after fault 12 is displayed and the reconnection loop starts, when I connect over USB to clear the nodeDB the region is also UNSET.  clearing nodeDB un-wedges.

it almost feels like swap starvation or something.  or a broken flash controller load balancing algorithm, like the counterfeit SDcards that report 256gb but really they just overwrite preexisting blocks after they go over 2gb

esev, 2025-01-18:
I've been digging a little deeper into the InternalFileSystem implementation. I'm not sure it is setup to work best with LittleFS. Normally LittleFS should be able to detect a failed flash write/erase and recover from that. But I think the InternalFileSystem implementation breaks that assumption.

Reason: The LittleFS documentation specifies that `block_size` should match the size of an erasable block. That is so LittleFS can read the block contents after writing to ensure they were written correctly. InternalFileSystem configures the LittleFS block_size to 128 bytes, while the flash erasable size is 4096 bytes. So when/if a flash erase/write fails 4096 bytes of the filesystem could be lost, but LittleFS is only able to detect and recover 128 bytes from that block.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/4dcfa3b7960a6dafec6736a5dbea255b0946be83/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L35

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/4dcfa3b7960a6dafec6736a5dbea255b0946be83/libraries/InternalFileSytem/src/InternalFileSystem.cpp#L109-L113

https://github.com/littlefs-project/littlefs/blob/7e110b44c0e796dc56e2fe86587762d685653029/lfs.h#L146-L150

Am I reading this correctly?

todd-herbert, 2025-01-18:
I do remember seeing discussion about that over at https://github.com/meshtastic/firmware/issues/4447, so I think you're probably right on the money again.

esev, 2025-01-18:
Ha! Thank you for pointing me to that bug. Yes, that describes what I was seeing exactly.

NireBryce, 2025-01-19:
that would track with at least my issue.

the thing that made mine work was 
- first killing the nodeDB 
- changing options that cause a reboot (and presumably config write) 
- and pairing a Bluetooth device (presumably creating a new paired devices list) 

if it's clobbering 3968 bytes of neighboring files for every erased 128B block it makes sense that rewriting the files would fix it.  

(Is there any chance this issue leaves garbage file fragments on the filesystem that isn't being prompted to be erased? I'm not familiar with LittleFS)

esev, 2025-01-19:
> if it's clobbering 3968 bytes of neighboring files for every erased 128B block it makes sense that rewriting the files would fix it.

Just to clarify. I don't believe this is happening for **every** write. There are two scenarios where I think it seems likely though:

1. If a Bluetooth device happens to move out of range during a write, the write seems to fail.
2. If the devices loses power during a write.

> Is there any chance this issue leaves garbage file fragments on the filesystem that isn't being prompted to be erased? I'm not familiar with LittleFS

I had originally thought LittleFS might be losing blocks, and slowly decreasing its capacity over time when this happens. However, after digging a little deeper into LittleFS, I notice it doesn't keep a free or corrupt block list. I think that leads to two outcomes.

1. If LittleFS thinks garbage file fragments are part of an existing file/directory then it is likely LittleFS will hit an assertion. I commonly saw `head >= 2 && head <= lfs->cfg->block_count` being associated with this.
2. If LittleFS doesn't have any file/directory reference the garbage file fragments, then it may reuse & overwrite those fragments when it needs to allocate new space.

esev, 2025-01-20:
Just saw another error on my device:

```
INFO  | 21:59:52 8749 [Router] Adding node to database with 80 nodes and 110280 bytes free!
DEBUG | 21:59:52 8749 [Router] Update changed=1 user DocB Heltec /DocP, id=0x433b9c3c, channel=0
DEBUG | 21:59:52 8749 [Router] Node status update: 36 online, 80 total
DEBUG | 21:59:52 8749 [Router] Opening /prefs/db.proto, fullAtomic=0
INFO  | 21:59:52 8749 [Router] Save /prefs/db.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:19: Assert lfs_cache_read 3539208231
ERROR | 21:59:52 8749 [Router] LFS assert: block < lfs->cfg->block_count
ERROR | 21:59:52 8749 [Router] Tr
```

That last log `Tr` is likely supposed to be `Trying FSCom.format()`. Note that the format attempt will lock the device. This happens because `format` tries to grab a lock that is already held by the file system operation that failed.

https://github.com/meshtastic/Adafruit_nRF52_Arduino/blob/4f591d0f71f75e5128fab9dc42ac72f1696cf89f/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp#L99

I've been looking for ways to recover from that.

tavdog, 2025-01-20:
Yeah that format attempt should be taken out because it does not work.

markbirss, 2025-01-20:
> that would track with at least my issue.
>
> the thing that made mine work was
>
> * first killing the nodeDB
> * changing options that cause a reboot (and presumably config write)
> * and pairing a Bluetooth [...]

You could try out "list-files" uf2 much same as nrf52-erase

https://discord.com/channels/867578229534359593/919642584480112750/1305904252626927729

[image]

Too confirm a erase it should list no files

[image]

esev, 2025-01-20:
> Yeah that format attempt should be taken out because it does not work.

I moved the format attempt so it happens after a reboot in #5900. I didn't want to remove it completely, as it'll help users recover from this filesystem corruption issue.

thebentern, 2025-01-20:
> > Yeah that format attempt should be taken out because it does not work.
> 
> I moved the format attempt so it happens after a reboot in [#5900](https://github.com/meshtastic/firmware/pull/5900). I didn't want to remove it completely, as it'll help users recover from this filesystem corruption issue.

@tavdog the internal lock Arduino was placing on the file system was why you were only able to get it working by modifying the code to format in the setup method

esev, 2025-01-21:
The reboot & format works. But I'm still seeing some FS corruption even with the retries. I'll keep digging into this.

```
INFO  | 02:36:30 35671 [Router] Node database full with 80 nodes and 124884 bytes free. Erasing oldest entry
INFO  | 02:36:30 35671 [Router] Adding node to database with 80 nodes and 124884 bytes free!
DEBUG | 02:36:30 35671 [Router] Update changed=1 user Tekify Fiber & Wireless/TEK, id=0x384a22ab, channel=0
DEBUG | 02:36:30 35671 [Router] Node status update: 49 online, 80 total
DEBUG | 02:36:30 35671 [Router] Opening /prefs/db.proto, fullAtomic=0
INFO  | 02:36:30 35671 [Router] Save /prefs/db.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:19: Assert lfs_cache_read 68842597
ERROR | 02:36:31 35672 [Router] LittleFS corruption detected: block < lfs->cfg->block_count
INFO  | 02:36:31 35672 [Router] Rebooting to format LittleFS
INFO  | 02:36:31 35672 [Router] NRF_POWER=0x40000000
<94><C3>^B@^AINFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 1 LittleFS format complete; restoring default settings
DEBUG | ??:??:?? 1 Filesystem files:
```

todd-herbert, 2025-01-21:
Do you have any idea as to the source of the corruption?

esev, 2025-01-21:
Not at all right now. :(

NireBryce, 2025-01-21:
> `INFO  | 02:36:30 35671 [Router] Node database full with 80 nodes and 124884 bytes free. Erasing oldest entry`
> `INFO  | 02:36:30 35671 [Router] Adding node to database with 80 nodes and 124884 bytes free!`

does it still corrupt if you start with an empty nodeDB?

esev, 2025-01-21:
> > `INFO  | 02:36:30 35671 [Router] Node database full with 80 nodes and 124884 bytes free. Erasing oldest entry`
> > `INFO  | 02:36:30 35671 [Router] Adding node to database with 80 nodes and 124884 bytes free!`
> 
> does it still corrupt if you start with an empty nodeDB?

Unfortunately, yes. I had been testing the format logic earlier in the day. So it did start from 0 at some point.

esev, 2025-01-21:
> [@esev](https://github.com/esev) [@todd-herbert](https://github.com/todd-herbert) I wonder if we should revisit the seek(0) and truncate approach and switch to more destructive delete, then open for write? We may have introduced a variable here which is muddying the water on the problem.

I'll create a build for testing just prior to the changes to SafeFile, and cherry-pick in our retry/format changes too.

One theory that I have is that maybe the flash writes are reporting success but a bit or two didn't actually get set correctly. IIUC the flash cache layer should still have a copy of the _expected_ write. I'll add some logic to compare it to what was _actually_ written.

LittleFS does this comparison too. But I believe LittleFS is only configured to look at the change of a single 128 byte block. Whereas the flash writes cover an entire 4096 byte flash page.

Mictronics, 2025-01-22:
I have seen a similar problem on a T-Echo.  Seen several fails on one of my T-Echos cause obviously by NodeDB file corruption.

During fail the log after reset is like this:
```
INFO | ??:??:?? 0

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  adafruit (directory)
DEBUG | ??:??:?? 0  bond_prph (directory)
DEBUG | ??:??:?? 0  bond_cntr (directory)
DEBUG | ??:??:?? 0  prefs (directory)
DEBUG | ??:??:?? 0  module.proto (113 Bytes)
DEBUG | ??:??:?? 0  channels.proto (102 Bytes)
DEBUG | ??:??:?? 0  config.proto (230 Bytes)
DEBUG | ??:??:?? 0  db.proto (8776 Bytes)
DEBUG | ??:??:?? 0  meshtastic.txt (342 Bytes)
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Use analog input 4 for battery level
INFO  | ??:??:?? 0 Scan for i2c devices
DEBUG | ??:??:?? 0 Scan for I2C devices on port 1
INFO  | ??:??:?? 0 PCF8563 found at address 0x51
INFO  | ??:??:?? 0 BME280 found at address 0x77
INFO  | ??:??:?? 0 2 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:7,2.5.20.22544c18c
DEBUG | ??:??:?? 0 Reset reason: 0x0
DEBUG | ??:??:?? 0 Set random seed 3136426161
INFO  | ??:??:?? 0 Init NodeDB
INFO  | ??:??:?? 0 Load /prefs/db.proto
ERROR | ??:??:?? 0 LFS assert: head >= 2 && head <= lfs->cfg->block_count
```

I modified the code to trigger an assert intentionally in lfs_file_open. This results in the following log (two times device reset):
```
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  prefs (directory)
ERROR | ??:??:?? 0 LFS assert: false
ERROR | ??:??:??��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  prefs (directory)
ERROR | ??:??:?? 0 LFS assert: false
ERROR | ??:??:??��@INFO  | ??:??:?? 0 
``` 
Again, on assert the device is frozen/locked .

Next I commented the `FSCom.format()`in the lfs_assert function, resulting in the following log:
```
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  prefs (directory)
ERROR | ??:??:?? 0 LFS assert: false
ERROR | ??:??:?? 0 Trying FSCom.format()
DEBUG | ??:??:?? 0    config.proto (272 Bytes)
ERROR | ??:??:?? 0 LFS assert: false
ERROR | ??:??:?? 0 Trying FSCom.format()
DEBUG | ??:??:?? 0    channels.proto (104 Bytes)
ERROR | ??:??:?? 0 LFS assert: false
ERROR | ??:??:?? 0 Trying FSCom.format()
DEBUG | ??:??:?? 0    db.proto (5938 Bytes)
ERROR | ??:??:?? 0 LFS assert: false
[... 107 lines cut ...]
INFO  | ??:??:?? 14 Save /prefs/db.proto
ERROR | ??:??:?? 14 Error: can't encode protobuf io error
ERROR | ??:??:?? 14 Can't write prefs!
ERROR | ??:??:?? 14 Failed to save to disk, retrying
DEBUG | ??:??:?? 15 Opening /prefs/config.proto, fullAtomic=1
ERROR | ??:??:?? 15 LFS assert: false
ERROR | ??:??:?? 15 Trying FSCom.format()
INFO  | ??:??:?? 15 Save /prefs/config.proto
ERROR | ??:??:?? 15 Error: can't encode protobuf io error
ERROR | ??:??:?? 15 Can't write prefs!
DEBUG | ??:??:?? 15 Opening /prefs/channels.proto, fullAtomic=1
ERROR | ??:??:?? 15 LFS assert: false
ERROR | ??:??:?? 15 Trying FSCom.format()
INFO  | ??:??:?? 16 Save /prefs/channels.proto
ERROR | ??:??:?? 16 Error: can't encode protobuf io error
ERROR | ??:??:?? 16 Can't write prefs!
DEBUG | ??:??:?? 16 Opening /prefs/db.proto, fullAtomic=0
ERROR | ??:??:?? 16 LFS assert: false
ERROR | ??:??:?? 16 Trying FSCom.format()
INFO  | ??:??:?? 16 Save /prefs/db.proto
ERROR | ??:??:?? 16 Error: can't encode protobuf io error
ERROR | ??:??:?? 16 Can't write prefs!
ERROR | ??:??:?? 16 NOTE! Record critical error 13 at src/mesh/NodeDB.cpp:1242
```
So it seems the issue is with infinitely blocked _lockFS.

esev, 2025-01-22:
Thanks @Mictronics for that very detailed analysis!

Would you be able to test with #5900?

Mictronics, 2025-01-22:
Sure, let me pull this in and I will report back.

Mictronics, 2025-01-22:
And another real test of the new fix: My mobile T-Echo failed again. :-( Again during a time when a lot of new nodes were coming in.

Device frozen in lockFS(), just after reset:
```

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  prefs (directory)
DEBUG | ??:??:?? 0    config.proto (270 Bytes)
DEBUG | ??:??:?? 0    channels.proto (277 Bytes)
DEBUG | ??:??:?? 0    db.proto (9504 Bytes)
DEBUG | ??:??:?? 0    module.proto (120 Bytes)
DEBUG | ??:??:?? 0  adafruit (directory)
DEBUG | ??:??:?? 0  bond_prph (directory)
DEBUG | ??:??:?? 0  bond_cntr (directory)
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Use analog input 4 for battery level
INFO  | ??:??:?? 0 Scan for i2c devices
DEBUG | ??:??:?? 0 Scan for I2C devices on port 1
INFO  | ??:??:?? 0 PCF8563 found at address 0x51
INFO  | ??:??:?? 0 BME280 found at address 0x77
INFO  | ??:??:?? 0 2 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:7,2.5.20.f21cae3cb
DEBUG | ??:??:?? 0 Reset reason: 0x0
DEBUG | ??:??:?? 0 Set random seed 3700966679
INFO  | ??:??:?? 0 Init NodeDB
INFO  | ??:??:?? 0 Load /prefs/db.proto
ERROR | ??:??:?? 0 LFS assert: head >= 2 && head <= lfs->cfg->block_count
```
Installed right away the build with latest HEAD up to 8e8b22e.
```
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  prefs (directory)
DEBUG | ??:??:?? 0    config.proto (270 Bytes)
DEBUG | ??:??:?? 0    channels.proto (277 Bytes)
DEBUG | ??:??:?? 0    db.proto (9504 Bytes)
DEBUG | ??:??:?? 0    module.proto (120 Bytes)
DEBUG | ??:??:?? 0  adafruit (directory)
DEBUG | ??:??:?? 0  bond_prph (directory)
DEBUG | ??:??:?? 0  bond_cntr (directory)
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Use analog input 4 for battery level
INFO  | ??:??:?? 0 Scan for i2c devices
DEBUG | ??:??:?? 0 Scan for I2C devices on port 1
INFO  | ??:??:?? 0 PCF8563 found at address 0x51
INFO  | ??:??:?? 0 BME280 found at address 0x77
INFO  | ??:??:?? 0 2 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 S:B:7,2.5.20.b8592f886
DEBUG | ??:??:?? 0 Reset reason: 0x0
DEBUG | ??:??:?? 0 Set random seed 2077297047
INFO  | ??:??:?? 0 Init NodeDB
INFO  | ??:??:?? 0 Load /prefs/db.proto
ERROR | ??:??:?? 0 LittleFS corruption detected: head >= 2 && head <= lfs->cfg->block_count
INFO  | ??:??:?? 0 Rebooting to format LittleFS
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 LittleFS format complete; restoring default settings
DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0 Power::lipoInit lipo sensor is not ready yet
DEBUG | ??:??:?? 0 Use analog input 4 for battery level
INFO  | ??:??:?? 0 Scan for i2c devices
[... 49 lines cut ...]
INFO  | 22:12:41 2 channel_num: 1
INFO  | 22:12:41 2 frequency: 869.525024
INFO  | 22:12:41 2 Slot time: 77 msec
INFO  | 22:12:41 2 Set radio: final power level=22
INFO  | 22:12:41 2 SX126x init result 0
INFO  | 22:12:41 2 Frequency set to 869.525024
INFO  | 22:12:41 2 Bandwidth set to 250.000000
INFO  | 22:12:41 2 Power output set to 22
DEBUG | 22:12:41 2 Current limit set to 140.000000
DEBUG | 22:12:41 2 Current limit set result 0
DEBUG | 22:12:41 2 Set DIO2 as RF switch, result: 0
DEBUG | 22:12:41 2 SX126X_RXEN not defined, defaulting to RADIOLIB_NC
DEBUG | 22:12:41 2 SX126X_TXEN not defined, defaulting to RADIOLIB_NC
DEBUG | 22:12:41 2 Use MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | 22:12:41 2 Set RX gain to boosted mode; result: 0
INFO  | 22:12:41 2 SX1262 init success
DEBUG | 22:12:41 2 LoRA bitrate = 118.394310 bytes / sec
INFO  | 22:12:41 2 PowerFSM init, USB power=1
DEBUG | 22:12:41 2 State: BOOT
ERROR | 22:12:41 2 NOTE! Record critical error 13 at src/platform/nrf52/main-nrf52.cpp:146
DEBUG | 22:12:41 2 [Power] Battery: usbPower=1, isCharging=1, batMv=4551, batPct=100
DEBUG | 22:12:41 2 [GPS] Probe for GPS at 9600
```

So while the actual fix is obviously solving the lock on boot issue the root cause of the NodeDB corruption issue still seem to exist.

esev, 2025-01-22:
That's a good test. Thanks for that!

https://github.com/meshtastic/Adafruit_nRF52_Arduino/pull/1 goes a long way toward addressing the corruption issue.

Mictronics, 2025-01-22:
Oh wow, caused be BT? I guess I can setup a long term log an play around here. I have two T-Echo on the bench I use for handheld purpose.
And yeah, the affected T-Echo is almost all time connected via BT to the phone when at home.

esev, 2025-01-22:
> Oh wow, caused be BT?

Seems to be. It's a really odd/unexpected interaction between the two.

> I guess I can setup a long term log an play around here. 

Thank you so much. This would be really helpful.

I've had mine crash a couple times still - but I can't rule out my own testing interfering with something. So far no issues in the last ~20 hours on my latest run.

I also added a memcmp in meshtastic/Adafruit_nRF52_Arduino#1 in the loops where the retries are handled. If the memory doesn't read back correctly it also retries. I'm not 100% sure that is needed though - as I haven't hit any corruption since running a build with that extra logic in it.

Mictronics, 2025-01-22:
But there is a nuance here. I had four different T-Echo failing this week. Out of this four, two were setup in the wild, a third one in the backyard. These three, I am 100% sure, had and still have BT disabled. Only the one handheld, that failed again today, has BT enabled.
So BT can't be the only cause for the flash corruption issue. I experience this issue only with a lot of node cycling through DB.

esev, 2025-01-22:
What firmware version were you running on them all?

esev, 2025-01-23:
I just got another crash as well. And I hadn't been home, connected to the node for 8+ hours when it crashed. None of the memcmp verification checks I added were ever triggered either.

This still wasn't pre-SafeFile changes though. That's my next test and will start running that code tonight.

Mictronics, 2025-01-23:
> This is a good data point. What firmware version were you running on them all?

Out in the wild something like 2.5.15 not sure. In the backyard 2.5.18 and the handheld one always the latest so 2.5.19 to 2.5.20. All custom build.

esev, 2025-01-23:
Here are the logs covering my previous 3 crashes.

```
DEBUG | 02:36:30 35671 [Router] Opening /prefs/db.proto, fullAtomic=0
INFO  | 02:36:30 35671 [Router] Save /prefs/db.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:19: Assert lfs_cache_read 68842597
ERROR | 02:36:31 35672 [Router] LittleFS corruption detected: block < lfs->cfg->block_count
```

```
DEBUG | 16:06:30 47125 [Router] Opening /prefs/db.proto, fullAtomic=0
INFO  | 16:06:30 47125 [Router] Save /prefs/db.proto
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS#
lfs debug:19: Assert lfs_cache_read 2411999021
ERROR | 16:06:30 47126 [Router] LittleFS corruption detected: block < lfs->cfg->block_count
```

```
DEBUG | 23:29:22 81090 [Router] Opening /prefs/db.proto, fullAtomic=0
INFO  | 23:29:22 81090 [Router] Save /prefs/db.proto
lfs debug:19: Assert lfs_cache_read 1212109790
ERROR | 23:29:22 81090 [Router] LittleFS corruption detected: block < lfs->cfg->block_count
```

Mictronics, 2025-01-23:
My log is running. Let's what we get. But my guess is a similar or equal error message.

esev, 2025-01-23:
> [@esev](https://github.com/esev) [@todd-herbert](https://github.com/todd-herbert) I wonder if we should revisit the seek(0) and truncate approach and switch to more destructive delete, then open for write? We may have introduced a variable here which is muddying the water on the problem.

This is worth investigating: https://github.com/littlefs-project/littlefs/pull/800 / https://github.com/littlefs-project/littlefs/issues/268

todd-herbert, 2025-01-23:
> Here are the logs covering my previous 3 crashes. I'm now running a build without [85de193](https://github.com/meshtastic/firmware/commit/85de193845bae37e1a19a87372e8c3d187f0ecb2) & [1c0f43c](https://github.com/meshtastic/firmware/commit/1c0f43c8e2fb4cbbc285eeac319d339ab63c866f)

Just to clarify, are those errors from a build before the switch to `truncate`?

esev, 2025-01-23:
Those were all with 85de193 & 1c0f43c.

todd-herbert, 2025-01-23:
> Those logs were all with [85de193](https://github.com/meshtastic/firmware/commit/85de193845bae37e1a19a87372e8c3d187f0ecb2) & [1c0f43c](https://github.com/meshtastic/firmware/commit/1c0f43c8e2fb4cbbc285eeac319d339ab63c866f).

Ah right! Fingers crossed that reverting them does help. I wouldn't be surprised, and at least it would point at a reasonable workaround for the time being.

mundeczek, 2025-01-24:
RAK4631, firmware 2.5.18.89ebafc, BT is not responding, log after connected to PC
```
”Ă..@..[32m.INFO  .[0m| ??:??:?? 0 .[32m.

//\ E S H T /\ S T / C

.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Filesystem files:
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m. adafruit (directory)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m. bond_prph (directory)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m. bond_cntr (directory)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.   RAK (108 Bytes)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m. prefs (directory)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.   module.proto (102 Bytes)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.   config.proto (177 Bytes)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.   channels.proto (106 Bytes)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.   db.proto.tmp (10392 Bytes)
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Power::lipoInit lipo sensor is not ready yet
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Use analog input 5 for battery level
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Scan for i2c devices
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Scan for I2C devices on port 1
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.SSD1306 found at address 0x3c
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.SSD1306 found at address 0x3c
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.0x3 subtype probed in 2 tries 
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Register MFG_UID FE: 0xffff
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Register MFG_UID 05: 0xffff
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.1 I2C devices found
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.acc_info = 0
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.S:B:9,2.5.18.89ebafc
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Build timestamp: 1735423672
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Reset reason: 0x0
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Set random seed 1743193640
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Init NodeDB
.[0m..[31m.ERROR .[0m| ??:??:?? 0 .[31m.Could not open / read /prefs/db.proto
.[0m..[33m.WARN  .[0m| ??:??:?? 0 .[33m.Devicestate 0 is old, discard
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Install default DeviceState
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Initial packet id 1194817520
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Partially randomized packet id 2537854961
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Use nodenum 0xd9856dfe 
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Load /prefs/config.proto
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded /prefs/config.proto successfully
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded saved config version 23
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Load /prefs/module.proto
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded /prefs/module.proto successfully
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded saved moduleConfig version 23
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Load /prefs/channels.proto
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded /prefs/channels.proto successfully
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Loaded saved channelFile version 23
.[0m..[31m.ERROR .[0m| ??:??:?? 0 .[31m.Could not open / read /prefs/uiconfig.proto
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.cleanupMeshDB purged 0 entries
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Use nodenum 0xd9856dfe 
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Adding node to database with 1 nodes and 154836 bytes free!
.[0m..[34m.DEBUG .[0m| ??:??:?? 0 .[34m.Expand short PSK #1
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Wanted region 3, using EU_868
.[0m..[32m.INFO  .[0m| ??:??:?? 0 .[32m.Save /prefs/db.proto
.[0m..[31m.ERROR .[0m| ??:??:?? 0 .[31m.LFS assert: block < lfs->cfg->block_count
.[0m..[31m.ERROR .[0m| ??:??:?? 0 .[31m.Trying FSCom.format()
.[0m.

JimTheCowboy, 2025-01-26:
> A couple of changes have been made which hopefully will fix this with the 2.5.20 release.

I've got my T-Echo up and running for about 3 hours on 2.5.20.4c97351 Alpha
NodeDB full (80+) so far I've logged 50+ BLE disconnects/connects without any issues. 

*keeping fingers crossed*

T-echo, 2.5.20 alpha FW, Android 13, App 2.5.16

[0m [32m INFO  [0m| 13:39:21 13076 [Router] [32m Node database full with 80 nodes and 114380 bytes free. Erasing oldest entry
[0m [32m INFO  [0m| 13:39:21 13076 [Router] [32m Adding node to database with 80 nodes and 114380 bytes free!

JimTheCowboy, 2025-01-26:
> I've got my T-Echo up and running for about 3 hours on 2.5.20.4c97351 Alpha NodeDB full (80+) so far I've logged 50+ BLE disconnects/connects without any issues.
>
> _keeping fingers crossed_
>
> T-echo, [...]

uptime 7h10mins, another 10+ deliberate connects/disconnects - no issues so far. 

IRL testing withouth logs will begin tomorrow.
hopefully it holds up!

very good job and many thanks to all contributors!

mb1986, 2025-01-26:
I can also confirm that everything seems to be fine now, thanks!

- hardware: rak4631
- nodes count: 80
- ble reconnects: over 15
- firmware: 2.5.20
- role: client

fifieldt, 2025-01-27:
Very many thanks to all contributors - whether you reported the bug, sent logs, tested firmware, read or wrote code, or kept your community up to date. Based on all reports so far, it does seem like this was solved in release 2.5.20. 

_A [full flash erase](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/) may be required if you are experiencing LFS assert issues._ [after upgrading to 2.5.20]

We'll soon mark this as closed by https://github.com/meshtastic/firmware/pull/5916

[2 more replies]

closed 2025-01-29 by fifieldt