meshtastic/firmware #5850: [Bug]: Remote Admin Shutdown Does Not Execute

IcyMidnight, 2025-01-13:
### Category

Other

### Hardware

T-Echo, Station G2

### Firmware Version

2.5.15.79da236

### Description

Using remote admin, I am able to change settings and reboot the other device reliably. Sending the shutdown command never has any effect, even if the command is confirmed delivered.

garthvh, 2025-01-13:
Where are you sending it from?

IcyMidnight, 2025-01-13:
This specific test:
Android app -IP-> Station G2 -RF-> T-Echo
They're 10 feet apart.

In the android app,
1. find my T-Echo in the node list
2. tap node id, `More details`
3. tap `Remote Administration`
4. tap `Shutdown`
5. "Delivery confirmed"

I think I also tested between my T-Echos and a T-Deck. All on 2.5.15. Never worked.