meshtastic/firmware #5865: [Bug]: CLIENT_MUTE never send NodeInfo packets

0x5c, 2025-01-16:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.14.f2ee0df and 2.5.18.89ebafc

### Description

When set to `CLIENT_MUTE`, my node never sends a nodeinfo packet. Whereas it reliably sends one within minutes of being set to `CLIENT`, and at every `nodeInfoBroadcastSecs`.

Tested on a Heltec Capsule Sensor v3, with `nodeInfoBroadcastSecs: 3600` (and also default), both left to its own for hours and with attempts at getting an early nodeinfo broadcast by having it forget a nearby active node.

### Relevant log output

```Shell

```

garthvh, 2025-01-16:
https://meshtastic.org/blog/choosing-the-right-device-role/#client-mute

0x5c, 2025-01-16:
Isn't NodeInfo something the node should be able to originate itself? To be clear, I am talking of the NodeInfo packet that allow *that* node to have others know it exist, not rebroadcasted NodeInfo messages.

As far as the docs say, it seems that the behaviour of not originating NodeInfo message is supposed to be specific to `CLIENT_HIDDEN` and `REPEATER`.

GUVWAF, 2025-01-16:
There is only one place in the code where `CLIENT_MUTE` is used (except for its definition) and that's at the place it decides to rebroadcast: https://github.com/search?q=repo%3Ameshtastic%2Ffirmware%20CLIENT_MUTE&type=code 
The setting doesn't affect NodeInfo broadcasts.

Are you sure you have other settings, like channel key and name, unchanged? Does normal messaging work?

0x5c, 2025-01-16:
> There is only one place in the code where CLIENT_MUTE is used (except for its definition) and that's at the place it decides to rebroadcast: https://github.com/search?q=repo%3Ameshtastic%2Ffirmware%20CLIENT_MUTE&type=code

Yeah I searched the code and couldn't see anything that could potentially have been relevant, glad to know I wasn't just lot there

> Are you sure you have other settings, like channel key and name, unchanged? Does normal messaging work?

The LoRa/etc settings weren't knowingly changed, and from comparing the settings of a different -perfectly working- node it seems they were indeed not changed.
Other than the lack of NodeInfo broadcast, most things work normally; default unchanged ch0 LongFast messages, traceroutes, receiving other node's NodeInfo messages.
Some things don't work, like power/environment telemetry broadcast, and sending DMs (that might be a key issue?), and of course the NodeInfo broadcasts

I have done a full factory-reset of the node, and it has not affected the persistence of the issue