meshtastic/firmware #5869: [Bug]: Changing Telemetry Values via Remote Admin resulted in new node ID for administered node

Deuteranomalous1, 2025-01-16:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.16

### Description

I was using remote administration via Android to change my telemetry intervals to prime numbers in an attempt to space out telemetry broadcasts as per discussion with b8b8 on our regional discord chat. Administering node was a T1000-e running 2.5.18. Administered node is RAK 4631 running 2.5.15 located on a mountain ~8km away and 800m higher than my house with good LOS. 

Changed Telemetry on 115503498 to 1801 1811 1823 1847 seconds respectively. 

The remote command went through just fine and then I immediately got a notification of a new node being discovered but it was the same name as the node I was administering. But this node has a different node number! New node number is 1323580256. 

Its the same node. All the settings I can check over remote admin are the same, the PKI admin keys still work just fine, for all I can see nothing has changed except the node name. Both old and new currently exist in my nodedb. 

here is the device metadata from the "new" node. 
PS C:\Users\XXXX> py -m meshtastic  --device-metadata --dest '!4ee43b60'
Connected to radio
INFO file:node.py getMetadata line:628 Requesting device metadata

firmware_version: 2.5.16.f81d3b0
device_state_version: 23

Sorry I can't seem to figure out how to export the debug log from android. If there is a way I am not aware of please LMK. In lieu of that I went through the logs and pulled out the relevant entries as screenshots. 

[image]
[image]
[image]
[image]

### Relevant log output

```Shell

```

Talie5in, 2025-01-16:
115503498 is a known NodeID from on RAK devices that is duplicated on many many many nodes - assumption is they image one device, then create a clone to mass-flash to additional ones after they have done testing on the first one to make sure it runs correctly.

Likely [PR#4562](https://github.com/meshtastic/firmware/issues/4559) has kicked in and regenerated your NodeID as the NodeID was also found in the NodeDB.

[image]
[image]

Deuteranomalous1, 2025-01-16:
> 115503498 is a known NodeID from on RAK devices that is duplicated on many many many nodes - assumption is they image one device, then create a clone to mass-flash to additional ones after they have done testing on the first one to make sure it runs correctly.
> 
> This is where a factory wipe and flashing fresh on a RAK on delivery is beneficial to avoid these cases.
> 
> Likely [PR#4562](https://github.com/meshtastic/firmware/issues/4559) has kicked in and regenerated your NodeID as the NodeID was also found in the NodeDB.
> 
> [image] [image]

Very interesting! Thank you I had no idea. I guess the reboot with the new telemetry settings triggered it to regenerate?

Deuteranomalous1, 2025-01-17:
Closed issue, I got my answer and it was my own negligence.

closed 2025-01-17 by Deuteranomalous1

Talie5in, 2025-01-17:
> I guess the reboot with the new telemetry settings triggered it to regenerate?

I believe that's what would of occurred yes.