meshtastic/firmware #5875: [Feature Request]: Add a client last connected timestamp to NODE_INFO

afourney, 2025-01-17:
### Platform

Cross-Platform

### Description

Often its unclear which nodes in a mesh are monitored, and which are infrastructure -- especially when the current recommendation is for most nodes to be client or client_mute.

This feature request is to add a timestamp to NODE_INFO, indicating when an interactive application was last connected to the node (via BLE, serial, or HTTPs). This could then be reflected in the node list (similar to last heard), so as to give some indication of which nodes are actively monitored.

thebentern, 2025-01-17:
I like this idea, but NodeInfo (specifically User) is something we have to guard pretty heavily against growth unless it's deemed a necessary addition, since it's such a prevalent packet on the mesh and also gets stored for each node on the device's constrained filesystem.

ctso, 2025-01-17:
What about a boolean on the `DeviceMetrics` proto?  `true` if something is connected via serial, tcp or bluetooth.  `false` otherwise.

afourney, 2025-01-17:
What about the device metrics packet? Those are more frequent, and it could just be one bit (is a client currently connected)?

swdrumm, 2025-02-17:
Jumping in here per GUVWAF's suggestion following [this conversation](https://discord.com/channels/867578229534359593/1338335272793800817) on Discord.  This timestamp functionality could be merged with my request for Read Receipts, as per below:

**Feature Request:** Show read receipts on the sender's node  when a direct message is read by the target node.

**Proposal: ** When the target node reads a DM, send a read receipt back to the sender. This should be disabled by default to minimize unnecessary network traffic.

GUVWAF, 2025-02-17:
For example, what could be done is send an additional bool with the ACK on a direct message that indicates whether there is currently a client connected to the node.

closed 2025-04-23 by github-actions[bot]