meshtastic/firmware #5876: [Bug]: T-Deck Bluetooth Won't Start w/ v2.5.19.f9876cf

peterhanneman, 2025-01-18:
### Category

BLE

### Hardware

T-Deck

### Firmware Version

2.5.19.f9876cf

### Description

I have two T-Decks that work as expected using v2.5.18 but on v2.5.19.f9876cf the Bluetooth functionality stops working and devices never show up in BLE scan.

Using "--set bluetooth.enabled true" does nothing and "--get bluetooth.enabled" returns false always.  Need to run "--set bluetooth.enabled true" after reflashing v2.5.18 to get it to work again.

### Relevant log output

```Shell

```

fifieldt, 2025-01-18:
Thanks for the report. A couple questions to help us understand:
1. Do you have wifi enabled?
2. What role is the device set to?

If you could connect it to a computer and get logs (eg click serial monitor on flasher.meshtastic.org ) that would be very helpful.

garthvh, 2025-01-18:
Are you using the new MUI?

peterhanneman, 2025-01-20:
1. WiFi was enabled on one T-Deck when it was flashed, but disabled on the other when flashed and both ended up in the same state.  I have tried to explicitly disable WiFi using "--set network.wifi_enabled false" before enabling Bluetooth but also didn't work.  Both devices seemed to function normally otherwise one with an uptime of ~55 hours.

2. One device has role set to Client and the other Client Mute.

Not using MUI with either of these currently just vanilla v2.5.19.f9876cf.

Can fetch logs tomorrow but I didn't see any obvious errors or warnings.

imp667, 2025-01-21:
I just got mine delivered today and got it setup with the app/Bluetooth then I decided to update the firmware. After that I am unable to get it to reconnect. The app does not even detect the device. I have also removed the Bluetooth device from the phone and removed the app. Reinstalled the app issue persists. Not a good first impression..

imp667, 2025-01-21:
> I just got mine delivered today and got it setup with the app/Bluetooth then I decided to update the firmware. After that I am unable to get it to reconnect. The app does not even detect the device. [...]

So I just used the website to try and enable Bluetooth but that did not work. Anyway to get the Bluetooth back on?

brvier, 2025-01-21:
> So I just used the website to try and enable Bluetooth but that did not work. Anyway to get the Bluetooth back on?

Rollback to the stable firmware. 2.18

peterhanneman, 2025-02-02:
I'm seeing the same error using Meshtastic v2.5.20.4c97351 Alpha.  Explicitly disabling WiFi before enabling Bluetooth using latest 2.5.11 CLI version still does not bring the Bluetooth radio up again.

I have observed the live logs using "meshtastic --monitor" and searching for bluetooth or radio related anything but see nothing.  If anyone wants to give me instructions on how to obtain more relevant detailed logs I'm happy to.

Besides the loss of Bluetooth functionality, v2.5.19.f9876cf Alpha is functioning stably on T-Deck for a full week of operation now.

Reverting to v2.5.18 Stable fixes the T-Deck Bluetooth regression.

meshtastic --info (partial, redacted)

`My info: { "myNodeNum": 0000000000, "rebootCount": 406, "minAppVersion": 30200, "deviceId": "pkey...==", "pioEnv": "t-deck" }
Metadata: { "firmwareVersion": "2.5.20.4c97351", "deviceStateVersion": 23, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "role": "CLIENT_MUTE", "positionFlags": 763, "hwModel": "T_DECK", "hasPKC": true, "excludedModules": 1280, "hasEthernet": false, "hasRemoteHardware": false }

Preferences: {
  "device": {
    "role": "CLIENT_MUTE",
    "nodeInfoBroadcastSecs": 3600,
    "tzdef": "EST5EDT,M3.2.0/2:00:00,M11.1.0/2:00:00",
    "ledHeartbeatDisabled": true,
    "serialEnabled": false,
    "buttonGpio": 0,
    "buzzerGpio": 0,
    "rebroadcastMode": "ALL",
    "doubleTapAsButtonPress": false,
    "isManaged": false,
    "disableTripleClick": false
  },
  "position": {
    "positionBroadcastSecs": 7200,
    "positionBroadcastSmartEnabled": true,
    "gpsUpdateInterval": 900,
    "positionFlags": 763,
    "broadcastSmartMinimumDistance": 125,
    "broadcastSmartMinimumIntervalSecs": 900,
    "gpsMode": "ENABLED",
    "fixedPosition": false,
    "gpsEnabled": false,
    "gpsAttemptTime": 0,
    "rxGpio": 0,
    "txGpio": 0,
    "gpsEnGpio": 0
  },
  "power": {
    "waitBluetoothSecs": 1,
    "lsSecs": 300,
    "minWakeSecs": 35,
    "isPowerSaving": false,
    "onBatteryShutdownAfterSecs": 0,
    "adcMultiplierOverride": 0.0,
    "sdsSecs": 0,
    "deviceBatteryInaAddress": 0,
    "powermonEnables": "0"
  },
  "network": {
    "wifiSsid": "<redacted>",
    "wifiPsk": "<redacted>",
    "wifiEnabled": false,
    "ntpServer": "",
    "ethEnabled": false,
    "addressMode": "DHCP",
    "rsyslogServer": "",
    "enabledProtocols": 0
  },
  "display": {
    "screenOnSecs": 600,
    "compassNorthTop": true,
    "gpsFormat": "DEC",
    "autoScreenCarouselSecs": 0,
    "flipScreen": false,
    "units": "METRIC",
[... 148 lines cut ...]
    "current": 10,
    "red": 191,
    "green": 147,
    "blue": 108,
    "ledState": false
  },
  "detectionSensor": {
    "minimumBroadcastSecs": 45,
    "detectionTriggerType": "LOGIC_HIGH",
    "enabled": false,
    "stateBroadcastSecs": 0,
    "sendBell": false,
    "name": "",
    "monitorPin": 0,
    "usePullup": false
  },
  "paxcounter": {
    "enabled": false,
    "paxcounterUpdateInterval": 0,
    "wifiThreshold": 0,
    "bleThreshold": 0
  },
  "version": 0
}
`

garthvh, 2025-02-02:
You want to disable power saving which is enabled by default on the tdeck to use Bluetooth

garthvh, 2025-02-02:
> I'm seeing the same error using Meshtastic v2.5.20.4c97351 Alpha. Explicitly disabling WiFi before enabling Bluetooth using latest 2.5.11 CLI version still does not bring the Bluetooth radio up again.
> 
> I have observed the live logs using "meshtastic --monitor" and searching for bluetooth or radio related anything but see nothing. If anyone wants to give me instructions on how to obtain more relevant detailed logs I'm happy to.
> 
> Besides the loss of Bluetooth functionality, v2.5.19.f9876cf Alpha is functioning stably on T-Deck for a full week of operation now.
> 
> Reverting to v2.5.18 Stable fixes the T-Deck Bluetooth regression.
> 
> meshtastic --info (partial, redacted)
> 
> ```
> Metadata: { "firmwareVersion": "2.5.20.4c97351", "deviceStateVersion": 23, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "role": "CLIENT_MUTE", "positionFlags": 763, "hwModel": "T_DECK", "hasPKC": true, "excludedModules": 1280, "hasEthernet": false, "hasRemoteHardware": false }
> 
> Preferences: {
>   "device": {
>     "role": "CLIENT_MUTE",
>     "nodeInfoBroadcastSecs": 3600,
>     "tzdef": "EST5EDT,M3.2.0/2:00:00,M11.1.0/2:00:00",
>     "ledHeartbeatDisabled": true,
>     "serialEnabled": false,
>     "buttonGpio": 0,
>     "buzzerGpio": 0,
>     "rebroadcastMode": "ALL",
>     "doubleTapAsButtonPress": false,
>     "isManaged": false,
>     "disableTripleClick": false
>   },
>   "position": {
>     "positionBroadcastSecs": 7200,
>     "positionBroadcastSmartEnabled": true,
>     "gpsUpdateInterval": 900,
>     "positionFlags": 763,
>     "broadcastSmartMinimumDistance": 125,
>     "broadcastSmartMinimumIntervalSecs": 900,
>     "gpsMode": "ENABLED",
>     "fixedPosition": false,
>     "gpsEnabled": false,
>     "gpsAttemptTime": 0,
>     "rxGpio": 0,
>     "txGpio": 0,
>     "gpsEnGpio": 0
>   },
>   "power": {
>     "waitBluetoothSecs": 1,
>     "lsSecs": 300,
>     "minWakeSecs": 35,
>     "isPowerSaving": false,
>     "onBatteryShutdownAfterSecs": 0,
>     "adcMultiplierOverride": 0.0,
>     "sdsSecs": 0,
>     "deviceBatteryInaAddress": 0,
>     "powermonEnables": "0"
>   },
>   "network": {
>     "wifiSsid": "<redacted>",
>     "wifiPsk": "<redacted>",
>     "wifiEnabled": false,
>     "ntpServer": "",
>     "ethEnabled": false,
>     "addressMode": "DHCP",
>     "rsyslogServer": "",
>     "enabledProtocols": 0
>   },
>   "display": {
>     "screenOnSecs": 600,
>     "compassNorthTop": true,
>     "gpsFormat": "DEC",
>     "autoScreenCarouselSecs": 0,
>     "flipScreen": false,
>     "units": "METRIC",
[... 150 lines cut ...]
>     "green": 147,
>     "blue": 108,
>     "ledState": false
>   },
>   "detectionSensor": {
>     "minimumBroadcastSecs": 45,
>     "detectionTriggerType": "LOGIC_HIGH",
>     "enabled": false,
>     "stateBroadcastSecs": 0,
>     "sendBell": false,
>     "name": "",
>     "monitorPin": 0,
>     "usePullup": false
>   },
>   "paxcounter": {
>     "enabled": false,
>     "paxcounterUpdateInterval": 0,
>     "wifiThreshold": 0,
>     "bleThreshold": 0
>   },
>   "version": 0
> }```
> ```

Bluetooth is off on this node, you need to enable Bluetooth

dvanmatre, 2025-02-02:
> Are you using the new MUI?

They specifically said what version of the firmware he was using in his original post and it’s not the UI version. 

> You want to disable power saving which is enabled by default on the tdeck to use Bluetooth

Power saving is/was not enabled on my T-Deck and I had the same issue with Bluetooth being disabled by a firmware update, as described. 

> Bluetooth is off on this node, you need to enable Bluetooth

His logs are from the 2.5.20 firmware, which when installed on a T-Deck disables Bluetooth. That’s the whole point.

peterhanneman, 2025-02-02:
Changing nothing but the firmware leaving all configuration settings the same:

Bluetooth connectivity works on v2.5.18 Stable
Bluetooth connectivity does NOT work on v2.5.19 Alpha or v2.5.20 Alpha

Running these CLIs commands:
`meshtastic --set network.wifi_enabled false`
`meshtastic --set bluetooth.enabled true`

raises the error:

`raise MeshInterface.MeshInterfaceError(meshtastic.mesh_interface.MeshInterfaceError: Timed out waiting for connection completion`

FYI enabling WiFi works as expected:
`meshtastic --set network.wifi_enabled true`

mcumings, 2025-02-06:
Having also experienced this (and regressing my TDecks back to 2.5.18), I can confirm that on my nodes:
- Wifi was disabled
- Bluetooth was enabled
- Power saving was not in play

The logs seemed to indicate that the BLE subsystem was not initializing at all upon boot, when compared to the logs of 2.5.18.

returntrip, 2025-02-07:
Same issue here with 2.5.20 Beta. Was working as fine on 2.5.18

thebentern, 2025-02-07:
Close with #5998

closed 2025-02-07 by thebentern