meshtastic/firmware #5880: [Documentation Change Request]: Battery Voltage for Seeed Xiao ESP32S3 Monitor

nis0521, 2025-01-18:
### Request Details

Problem: Seeed Xiao Meshtastic boards don’t have a native way of measuring battery voltage when powered by batteries. 

Solution: Because the the ESP32S3 super small boards have been out for a while, there is a workaround where you use 2 resisters from the battery terminal to a a2d pin on the board and uses a few line of code. It’s actually documented on the official product page 

https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/

Issue: In order to make this work those lines of code need to make it into the firmware. I don’t know how I would go about doing this or this can be worked by the devs. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

caveman99, 2025-01-18:
The modification you quote is for the C3 board, not the S3 board. While there is a similar mod for the S3 board we won't add that to the stock firmware, as all of the available GPIO pins re tied up already. For measurements i recommend adding an INA2xx probe.

closed 2025-01-18 by caveman99

nis0521, 2025-01-22:
Okay thanks for looking into this. 

I think the main attraction for this board is the size. Adding to the board kind of defeats the purpose of the board. It’d be great if this reconsidered maybe when you guys have more bandwidth:(

Thinking about making a few remote gadgets. Not having battery monitoring is a big issue :(

nis0521, 2025-02-01:
@caveman99 so went and got some INA219 boards and an android to get this to work. Doesn't seem to be working Voltage and  Current comes in on Channel 3 and not on the Battery which is just at 0%. Channel seems to be correct at 0x40 (64). 

Any way you can reopen this as an active issue? Another dude seems like he has a workaround: 

https://github.com/meshtastic/firmware/issues/5248