meshtastic/firmware #5882: [Feature Request]: Support for RAK12044 Hall effect sensor

rjzak, 2025-01-18:
### Platform

NRF52

### Description

I'm very new to Meshtastic, so I apologize if this was mentioned elsewhere (I did search the issues and PRs), but it would be neat to support the Rak wireless RAK12044 Hall effect sensor to have a compass.

https://store.rakwireless.com/products/wisblock-hall-effect-magnetic-sensor-rak12044

I'm also assuming that since this device isn't listed under [Hardware -> Devices -> RAK -> Wisblock -> Peripherals](https://meshtastic.org/docs/hardware/devices/rak-wireless/wisblock/peripherals/) that it's not supported currently.

benkyd, 2025-03-26:
Happy to implement this - I do not have the hardware required though. Not sure what the process for sourcing unproven contributors hardware is on a temp basis.

rjzak, 2025-03-27:
The part is $8. I'll buy and send it to you if you'd like. Email me. richard.j.zak@gmail.com

benkyd, 2025-03-27:
> The part is $8. I'll buy and send it to you if you'd like. Email me. [richard.j.zak@gmail.com](mailto:richard.j.zak@gmail.com) or [rjzak@protonmail.ch](mailto:rjzak@protonmail.ch).

Very kind offer, i also don't have a wizblok / other RAK host. If nobody else picks up this issue i'll find a way to source one. Note i am in the U.K.

tropho23, 2025-04-06:
Unfortunately this hall effect sensor will not be sensitive enough to serve as a compass component; it's sensitivity is limited to 15 mV/mT, and the Earth's magnetic field fluctuates at a tiny fraction of that relatively huge unit of measurement.  You also must pair magnetometers with accelerometers and/or gyroscopes to serve effectively as electronic compasses.  The DRV5056A4QDBZR chip used is really only for detecting magnetic fields in the immediate area, such as for mobile robots, detecting ferrous metals, etc.

Something like the AK09911C would be more useful, which has sensitivity as granular as 0.6 μ T/LSB.

But there is no need for that; the RAK12034 9-axis IMU includes magnetometer, accelerometer, and gyroscope sensors to serve as a compass and perhaps more.  It's already supported by Meshtastic: https://store.rakwireless.com/products/rak12034-9-axis-accelerometer

The RAK12034 is $12.50 USD vs $8 but it works today without issue.  I have purchased several and can attest they work well.

closed 2025-04-06 by garthvh