meshtastic/firmware #5884: [Bug]: webserver returns `404` exclusively

kquinsland, 2025-01-19:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.18.89ebafc

### Description

My system is a brand new raspberry pi os (lite edition, based on bookworm) running on the Pi Zero 2 W.
I am using the Waveshare SX1262 hat with the L76K GNSS module.
I was able to get the thing [booting](https://github.com/meshtastic/firmware/issues/5883) and now I am unable to access anything via the web UI.

The problem:

```shell
root@meshtastic:/usr/share/doc/meshtasticd/web# journalctl -xfeu meshtasticd.service
<...>
Jan 18 19:02:10 mesh-node meshtasticd[1011]: INFO  | 03:02:10 0 SX1262 init success
Jan 18 19:02:10 mesh-node meshtasticd[1011]: ERROR | 03:02:10 0 Error reading File : certificate.pem
Jan 18 19:04:55 mesh-node meshtasticd[1011]: ERROR | 03:02:10 0 ERROR SSL Certificate File can't be loaded or is missing
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 Create SSL Cert -certificate.pem- succesfull
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 Use webserver port from yaml config 443
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 Webserver started
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 Web Server framework started on port: 443
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 Web Server root /usr/share/doc/meshtasticd/web
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 API server listen on TCP port 4403
Jan 18 19:04:55 mesh-node meshtasticd[1011]: DEBUG | 03:02:13 3 LoRA bitrate = 118.394310 bytes / sec
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:13 3 PowerFSM init, USB power=1
Jan 18 19:04:55 mesh-node meshtasticd[1011]: DEBUG | 03:02:13 3 State: BOOT
Jan 18 19:04:55 mesh-node meshtasticd[1011]: INFO  | 03:02:40 30 [NodeInfo] Send our nodeinfo to mesh (wantReplies=1)
<...>
```

So let's go to that directory and see what's there:

```shell
root@meshtastic:/usr/share/doc/meshtasticd/web# ls
apple-touch-icon.png  favicon.ico  icon.svg  index.html  Logo_Black.svg  Logo_White.svg  robots.txt  site.webmanifest  static
```

And pick any file that the web server _should_ be able to return:

```shell
root@meshtastic:/usr/share/doc/meshtasticd/web# cat robots.txt 
User-agent: *
Allow: /
```

Ok, so let's try to get that:

```shell
root@meshtastic:/usr/share/doc/meshtasticd/web# curl --insecure -vvv https://localhost/robots.txt
*   Trying 127.0.0.1:443...
* Connected to localhost (127.0.0.1) port 443 (#0)
* ALPN: offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN: server accepted http/1.1
* Server certificate:
*  subject: C=DE; O=Meshtastic; CN=meshtastic.local
*  start date: Jan 19 03:02:14 2025 GMT
*  expire date: Jan 19 03:02:14 2026 GMT
*  issuer: C=DE; O=Meshtastic; CN=meshtastic.local
*  SSL certificate verify result: self-signed certificate (18), continuing anyway.
* using HTTP/1.1
> GET /robots.txt HTTP/1.1
> Host: localhost
> User-Agent: curl/7.88.1
> Accept: */*
> 
< HTTP/1.1 404 Not Found
< Date: Sun, 19 Jan 2025 03:10:25 GMT
< Access-Control-Allow-Origin: *
< Content-Length: 14
< 
* Connection #0 to host localhost left intact
File not found
```

Nothing shows up in the logs for this request so I don't know what's going on. Any ideas?

My system:

```shell
root@meshtastic:~# uname -a
Linux mesh-node 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
root@meshtastic:~# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

My config:

```shell
root@meshtastic:~# cat /etc/meshtasticd/config.yaml
---
Lora:
    Module: sx1262
    DIO2_AS_RF_SWITCH: true
    CS: 21
    IRQ: 16
    Busy: 20
    Reset: 18
    # This is new to me, came from the reference file at
    #   /etc/meshtasticd/available.d/lora-waveshare-sxxx.yaml
    SX126X_ANT_SW: 6

Webserver:
    Port: 443
    # Dir is created/populated when the deb file is installed
    RootPath: /usr/share/doc/meshtasticd/web

Logging:
    # TODO: revert back if I'm able to fix my issue!
    LogLevel: debug
```

### Relevant log output

```Shell

```

vidplace7, 2025-01-23:
@kquinsland Try changing
```
RootPath: /usr/share/doc/meshtasticd/web
```
to
```
RootPath: /usr/share/meshtasticd/web
```

The web directory was recently moved there, as it's a more appropriate location.
We attempted to symlink from `/usr/share/doc/meshtasticd/web` to `/usr/share/meshtasticd/web` in the packaging, but it doesn't appear to be working correctly for all users.

closed 2025-01-23 by vidplace7

DanceMore, 2025-03-23:
I am using the docker container [as described here](https://meshtastic.org/docs/software/linux/usage/).

I see log messages related to web server startup:

```
ERROR | 00:01:57 0 Error reading File : certificate.pem
ERROR | 00:01:57 0 ERROR SSL Certificate File can't be loaded or is missing
INFO  | 00:01:57 0 Create SSL Cert -certificate.pem- succesfull
INFO  | 00:01:57 0 Webserver port in yaml config set to 0, defaulting to port 443
INFO  | 00:01:57 0 Webserver started
INFO  | 00:01:57 0 Web Server framework started on port: 443
INFO  | 00:01:57 0 Web Server root
INFO  | 00:01:57 0 API server listen on TCP port 4403
```

I do not even have a `/usr/share/meshtasticd`.

```
root@0adb8d5e1b65:/# ls -l /usr/share
total 124
drwxr-xr-x  2 root root 4096 Feb  3 00:00 base-files
drwxr-xr-x  2 root root 4096 Feb  3 00:00 base-passwd
drwxr-xr-x  3 root root 4096 May 25  2023 bash-completion
drwxr-xr-x  4 root root 4096 Feb  3 00:00 bug
drwxr-xr-x  2 root root 4096 Feb  3 00:00 common-licenses
drwxr-xr-x  2 root root 4096 Feb  3 00:00 debconf
drwxr-xr-x  3 root root 4096 Feb  3 00:00 debianutils
drwxr-xr-x  2 root root 4096 Dec 31 10:25 dict
drwxr-xr-x  1 root root 4096 Feb  5 02:30 doc
drwxr-xr-x  2 root root 4096 Feb  3 00:00 doc-base
drwxr-xr-x  3 root root 4096 Feb  3 00:00 dpkg
drwxr-xr-x  3 root root 4096 Jan  8  2023 gcc
drwxr-xr-x  3 root root 4096 Jan  8  2023 gdb
drwxr-xr-x  2 root root 4096 Feb  3 00:00 info
drwxr-xr-x  2 root root 4096 Feb  3 00:00 keyrings
drwxr-xr-x  2 root root 4096 Feb  3 00:00 libc-bin
drwxr-xr-x  2 root root 4096 Feb  3 00:00 libgcrypt20
drwxr-xr-x  4 root root 4096 Feb  3 00:00 lintian
drwxr-xr-x  2 root root 4096 Feb  3 00:00 locale
drwxr-xr-x  6 root root 4096 Feb  3 00:00 man
drwxr-xr-x  2 root root 4096 Feb  3 00:00 menu
drwxr-xr-x  2 root root 4096 Dec 31 10:25 misc
drwxr-xr-x  2 root root 4096 Feb  3 00:00 pam
drwxr-xr-x  2 root root 4096 Feb  3 00:00 pam-configs
drwxr-xr-x  4 root root 4096 Feb  3 00:00 perl5
drwxr-xr-x  2 root root 4096 Feb  3 00:00 pixmaps
drwxr-xr-x  3 root root 4096 May 11  2023 polkit-1
drwxr-xr-x  2 root root 4096 Feb  3 00:00 tabset
drwxr-xr-x  2 root root 4096 May  7  2023 terminfo
drwxr-xr-x  3 root root 4096 Nov 21 20:01 util-linux
drwxr-xr-x 20 root root 4096 Feb  3 00:00 zoneinfo
```

I can't find a directory named `web/` anywhere inside the container.

kquinsland, 2025-03-23:
@DanceMore check `/etc/meshtasticd` for a config file.  A `cd /; sudo find . -name meshtasticd` might offer a clue to where things are in the container image

DanceMore, 2025-03-24:
I already went there, looking for a starter config file from which to modify or learn default values; and left even more confused.

```
% docker exec -it meshtasticd /bin/bash
root@33e9bb056bf4:/var/lib/meshtasticd# cd /etc/meshtasticd/
root@33e9bb056bf4:/etc/meshtasticd# ls
available.d  config.d
```

similarly, `find` was not very informative.

```
root@33e9bb056bf4:/etc/meshtasticd# cd /; sudo find . -name meshtasticd
bash: sudo: command not found
root@33e9bb056bf4:/# cd /; find . -name meshtasticd
./etc/meshtasticd
./var/lib/meshtasticd
./usr/sbin/meshtasticd
```

assume I know a lot about Docker and Linux but nothing about meshtastic (yet). it feels like this container assumes I have a pre-existing config to bring and/or other "meshtastic things", like my own opinions on the `web/` directory??

I have hardware in the mail and was trying to get ahead on the software side and I keep as much as I can inside Docker for workstation / development hygiene.

it does boot up `/usr/sbin/meshtasticd` just fine, complete with whatever "simulation" nodes it launches for testing if you have zero config or hardware specified. I wanted to see the Web UI and learn from the sim nodes before whatever my next steps are :)