meshtastic/firmware #5893: [Bug]: GPS not working on LILYGO T-Beam 1.2

PaoloneM, 2025-01-19:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.5.18.89ebafc Beta

### Description

Hello, I filled a bug since I wasn't able to find a more suitable issue type... Might be I simply need some support on warming up my board. I just received a brand new LILYGO T-Beam V1.2 ESP32 LoRa 433MHz and I'm not able to have the GPS module working. 

I left the board on open sky for a couple of hours but still no GPS signal. 

On the iOS app no position log is available:

[image]

Also tested firmware `2.5.19.f9876cf Alpha`

My board's config:

```
─➤  meshtastic --export-config
# start of Meshtastic configure yaml

config:
  bluetooth:
    enabled: true
    fixedPin: xxxxxx
    mode: FIXED_PIN
  device:
    nodeInfoBroadcastSecs: 3600
    role: CLIENT_MUTE
  display:
    screenOnSecs: 600
  lora:
    hopLimit: 3
    region: EU_433
    sx126xRxBoostedGain: true
    txEnabled: true
    txPower: 30
    usePreset: true
  network:
    ntpServer: meshtastic.pool.ntp.org
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsUpdateInterval: 15
    positionBroadcastSecs: 900
    positionBroadcastSmartEnabled: true
    positionFlags: 811
  power:
    lsSecs: 300
    minWakeSecs: 10
    sdsSecs: 4294967295
    waitBluetoothSecs: 60
module_config:
  ambientLighting:
    blue: 188
    current: 10
    green: 44
    red: 131
  detectionSensor:
    detectionTriggeredHigh: true
    minimumBroadcastSecs: 45
[... 291 lines cut ...]
INFO  | 21:18:31 28 [Router] Received routing from=0x1c2f13fc, id=0x15b5163d, portnum=3, payloadlen=23
DEBUG | 21:18:31 28 [Router] Routing sniffing (id=0x15b5163d fr=0x1c2f13fc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1737321511 rxSNR=12 rxRSSI=-72 hopStart=3)
DEBUG | 21:18:31 28 [Router] No rebroadcast: Role = CLIENT_MUTE or Rebroadcast Mode = NONE
DEBUG | 21:18:31 28 [Router] Delivering rx packet (id=0x15b5163d fr=0x1c2f13fc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1737321511 rxSNR=12 rxRSSI=-72 hopStart=3)
DEBUG | 21:18:31 28 [Router] Update DB node 0x1c2f13fc, rx_time=1737321511
DEBUG | 21:18:31 28 [Router] Forwarding to phone (id=0x15b5163d fr=0x1c2f13fc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1737321511 rxSNR=12 rxRSSI=-72 hopStart=3)
DEBUG | 21:18:31 28 [Router] Module 'routing' considered
INFO  | 21:18:31 28 Tell client we have new packets 4
INFO  | 21:18:31 28 BLE notify fromNum
DEBUG | 21:18:31 28 FromRadio=STATE_SEND_PACKETS
DEBUG | 21:18:31 28 phone downloaded packet (id=0x15b5163d fr=0x1c2f13fc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxtime=1737321511 rxSNR=12 rxRSSI=-72 hopStart=3)
INFO  | 21:18:34 31 [NodeInfo] Send our nodeinfo to mesh (wantReplies=1)
INFO  | 21:18:34 31 [NodeInfo] Send owner !bb832cbc/IU2RPI/R_T1
DEBUG | 21:18:34 31 [NodeInfo] Partially randomized packet id 3267963543
DEBUG | 21:18:34 31 [NodeInfo] Update DB node 0xbb832cbc, rx_time=1737321514
DEBUG | 21:18:34 31 [NodeInfo] handleReceived(LOCAL) (id=0xc2c92a97 fr=0xbb832cbc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1737321514 priority=10)
DEBUG | 21:18:34 31 [NodeInfo] No modules interested in portnum=4, src=LOCAL
DEBUG | 21:18:34 31 [NodeInfo] Add packet record (id=0xc2c92a97 fr=0xbb832cbc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1737321514 priority=10)
DEBUG | 21:18:34 31 [NodeInfo] Expand short PSK #0
DEBUG | 21:18:34 31 [NodeInfo] Use AES0 key!
DEBUG | 21:18:34 31 [NodeInfo] enqueue for send (id=0xc2c92a97 fr=0xbb832cbc to=0xffffffff, WantAck=0, HopLim=3 Ch=0x63 encrypted len=63 rxtime=1737321514 hopStart=3 priority=10)
DEBUG | 21:18:34 31 [NodeInfo] txGood=0,txRelay=0,rxGood=1,rxBad=0
INFO  | 21:18:34 31 Tell client we have new packets 5
INFO  | 21:18:34 31 BLE notify fromNum
```

fifieldt, 2025-01-19:
Thanks for writing in in such detail. I can see from the logs that your GPS was detected successfully as a Ublox 6. 

So, unless there's a bug, most likely a settings thing. Are you outdoor? How long did you wait?

PaoloneM, 2025-01-20:
HI, thnaks for you quick reply.

I left the bourd outdoor for 2 hour approximately on my car's dashboard, right behind the windshield. My car also has full glass ceiling, I generally have no issues with GPS devices in it. 

I'll try a longer time / different location.

fifieldt, 2025-01-20:
Ok, so reception is probably not an issue then. Will have to look at the settings!

PaoloneM, 2025-01-20:
Hi, the position was finally locked after several hours. It's not accurate and signal is lost easily, it's probably an antenna issue, I'll try a different one. I'll then review the config and the source code for a better understanding of the settings.

I'm going to close this issue, adding some info about current config and related log for reference and personal future investigation.

I updated the module conf withe the following:

```
gpsUpdateInterval: 15
positionBroadcastSecs: 30
```

and the corresponding relevant log is

```
DEBUG | ??:??:?? 10 [GPS] NMEA GPS time 2025-01-20 21:46:26 age 4
DEBUG | ??:??:?? 10 [GPS] Upgrade time to quality GPS
DEBUG | 21:46:26 10 [GPS] Read RTC time as 1737409586
DEBUG | 21:46:26 10 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:46:26 10 [GPS] onGPSChanged() pos@0 time=1737409586 lat=0 lon=0 alt=0
INFO  | 21:46:26 10 [GPS] updatePosition LOCAL pos@0 time=1737409586 lat=0 lon=0 alt=0
DEBUG | 21:46:26 10 [GPS] Set local position: lat=0 lon=0 time=1737409586 timestamp=0
DEBUG | 21:46:26 10 [GPS] Node status update: 1 online, 2 total

...

WARN  | 22:01:16 901 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time
DEBUG | 22:01:16 901 [GPS] Took 900s to get lock
DEBUG | 22:01:16 901 [GPS] Predict 0s to get next lock
DEBUG | 22:01:16 901 [GPS] 14s until next search
DEBUG | 22:01:16 901 [GPS] gps_update_interval >= 0s needed to justify hardsleep
INFO  | 22:01:17 901 [GPS] GPS power state move from ACTIVE to HARDSLEEP
DEBUG | 22:01:17 901 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 22:01:17 901 [GPS] onGPSChanged() pos@0 time=1737410477 lat=0 lon=0 alt=0
INFO  | 22:01:17 901 [GPS] updatePosition LOCAL pos@0 time=1737410477 lat=0 lon=0 alt=0
DEBUG | 22:01:17 901 [GPS] Set local position: lat=0 lon=0 time=1737410477 timestamp=0
DEBUG | 22:01:17 901 [GPS] Node status update: 1 online, 2 total
DEBUG | 22:01:17 902 [Power] Battery: usbPower=1, isCharging=1, batMv=4144, batPct=93
INFO  | 22:01:32 916 [GPS] GPS power state move from HARDSLEEP to ACTIVE

...

DEBUG | 22:06:48 1232 [Router] Set local position: lat=0 lon=0 time=1737410808 timestamp=0
WARN  | 22:06:48 1232 [Router] Skip position send because lat/lon are zero!
INFO  | 22:06:48 1232 [Router] Asked module 'position' to send a response
DEBUG | 22:06:48 1232 [Router] Module 'routing' wantsPacket=1

...

WARN  | 22:16:32 1816 [GPS] Couldn't publish a valid location: didn't get a GPS lock in time
DEBUG | 22:16:32 1816 [GPS] Took 900s to get lock
DEBUG | 22:16:32 1816 [GPS] Predict 900s to get next lock
DEBUG | 22:16:32 1816 [GPS] 0s until next search
INFO  | 22:16:32 1816 [GPS] GPS power state move from ACTIVE to IDLE
DEBUG | 22:16:32 1816 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 22:16:32 1816 [GPS] onGPSChanged() pos@0 time=1737411392 lat=0 lon=0 alt=0
INFO  | 22:16:32 1816 [GPS] updatePosition LOCAL pos@0 time=1737411392 lat=0 lon=0 alt=0
DEBUG | 22:16:32 1816 [GPS] Set local position: lat=0 lon=0 time=1737411392 timestamp=0
DEBUG | 22:16:32 1816 [GPS] Node status update: 2 online, 2 total
INFO  | 22:16:37 1821 [GPS] GPS power state move from IDLE to ACTIVE
```

Whit those settings I was expecting to see more activity (every 15" some GPS info, every 30" some radio acrivity to send current position even if not locked).

The [docs](https://meshtastic.org/docs/configuration/radio/position/#broadcast-interval) tell:

> The GPS updates will be sent out every broadcast interval, either with the actual GPS location or an empty location if no GPS fix was achieved.

While the log says

> WARN  | 22:06:48 1232 [Router] Skip position send because lat/lon are zero!

and this seems to be in conflict.

Thanks again for the support! 👋

closed 2025-01-20 by PaoloneM

fifieldt, 2025-01-20:
Thanks so much for the update. Feel free to open another issue if you need it!

fifieldt, 2025-01-20:
It might be the refresh times are too low, too.