meshtastic/firmware #5895: [Feature Request]: The device displays Chinese fonts.

lunneswang, 2025-01-20:
### Platform

NRF52

### Description

Excuse me, when can I display Chinese fonts on my device? At present, many China users are also using Meshtastic in different fields.

fifieldt, 2025-01-21:
Thanks for writing in. Definitely something we'd like if it's possible within the limited space in these devices. At the moment the only fonts available are hand-drawn glyphs, european languages with limited differences to English. It's a bit of technical research work that needs to be done to work out how to do it for CJK languages and if it's possible - help needed!

hayschan, 2025-01-21:
One of the [Meshtastic China Community](https://meshcn.net/)'s member, Mario @ChangYanChu, has been actively developing such feature - showing Chinese character on Meshtastic UI. It seems that it is way harder to do so on OLED screen than on TFT screen. 

Thanks to his dedication, he had some initial breakthrough today - showing the term `signal` in Simplified Chinese.

[image]

fifieldt, 2025-01-21:
太好了！

hayschan, 2025-01-21:
This is his commit for the code change to test Chinese font support: https://github.com/ChangYanChu/firmware/commit/e133484e1e6b942ff1eae1b8ea3d7d66f45cd6d9

@fifieldt can you please help to take a look at it? If possible, we would like to make it compatible with the master and merge Chinese font support into Meshtastic firmware repo. Let us know whether we are in the correct direction and in general any suggestion is welcomed.

https://github.com/user-attachments/assets/c017d6ea-0917-4f67-a74a-5f91c516787a