meshtastic/firmware #5896: [Bug]: FSCom.format() freezes device from lfs_assert

esev, 2025-01-20:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.5.19

### Description

This might impact all nrf52 devices

https://github.com/meshtastic/firmware/blob/950341d1f980369b58cc3d63dac6b036effc0270/src/FSCommon.cpp#L55-L68

lfs_assert is called from within LittleFS whenever it encounters an unrecoverable file system error. At the point where it is called, [Adafruit_LittleFS.cpp](https://github.com/meshtastic/Adafruit_nRF52_Arduino/blob/master/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp) has likely already acquired its `_mutex` via `_lockFS()`.

Calling format will attempt to acquire the lock again, likely causing deadlock.

https://github.com/meshtastic/Adafruit_nRF52_Arduino/blob/4f591d0f71f75e5128fab9dc42ac72f1696cf89f/libraries/Adafruit_LittleFS/src/Adafruit_LittleFS.cpp#L97-L99.

### Relevant log output

```Shell

```

todd-herbert, 2025-01-20:
Only an anecdote, but I have also definitely seen the firmware hang at `LOG_ERROR("Trying FSCom.format()");` before.

esev, 2025-01-20:
There is a hardware register (GPREGRET) that maintains its value across a soft reboot, My current plan is to modify lfs_assert to set this register to a magic value and then reboot. I'll check for this value in nrf52Setup, and if it exists, call FScom.format from there.

todd-herbert, 2025-01-20:
I'm not sure if it's a deal-breaker, but I'm pretty sure the bootloader does use that register too:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/3a2708b30a0d2c35a8162762512fa8bead311502/src/main.c#L98-L104

esev, 2025-01-20:
Thanks for that! I was wondering what other values existed. As long as I choose a value that doesn't conflict with those it works.

thebentern, 2025-01-20:
> Thanks for that! I was wondering what other values existed. As long as I choose a value that doesn't conflict with those it works.

For science! 🧑‍🔬

PR #5900 (2025-01-20): Reboot before formatting LittleFS
Fixes #5896
closed 2025-01-20 by thebentern