meshtastic/firmware #5904: [Bug]: Store&Forward not setup on Heltec E213 and T190 because PSRAM not found

NH-RedAnt, 2025-01-22:
### Category

Hardware Compatibility

### Hardware

Heltec Vision Master E213

### Firmware Version

2.5.18 and 2.5.20

### Description

Trying to test Store&Forward on a collection of Heltec nodes.  The E213 and T190 ESP32s3 hardware are supposed to have 8MB of PSRAM.  The serial logs report 0 PSRAM found, and disable S&F.  First tested with an image from the web installer, then a local PlatformIO build of 2.5.20 from the github repo.

First I thought that the firmware/variants/heltec_vision_master_e213/platformio.ini just required a additional "build_flags" of "-DBOARD_HAS_PSRAM".  But then discovered that was already being set in "firmware/boards/heltec_vision_master_e213.json".

Suspected it could be some other incorrectly set variable but have not found the correct combination yet. I've tried adding:

  build.extra_flags: 
      "-mfix-esp32-psram-cache-issue", 
      "-mfix-esp32-psram-cache-strategy=memw",

and 
    "build.flash_mode": "dio",
    "build.psram_type": "qspi",

Settings found on Heltec E213 example https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/boards.txt

Wondering if anyone has hints on where I might focus my efforts.  Is there anything special that needs to be done for PSRAM?  Any oddities with the PSRAM test code?  Are any other ESP32s3 MCU platforms working OK with S&F?

### Relevant log output

```Shell
[0m␀␛[32m␀INFO  ␛[0m| ??:??:?? 4 ␛[32m␀S:B:67,2.5.15.79da236
␛[0m␀␛[32m␀INFO  ␛[0m| ??:??:?? 4 ␛[32m␀Build timestamp: 1732889532
␛[0m␀␛[34m␀DEBUG ␛[0m| ??:??:?? 4 ␛[34m␀Total heap: 280536
␛[0m␀␛[34m␀DEBUG ␛[0m| ??:??:?? 4 ␛[34m␀Free heap: 238112
␛[0m␀␛[34m␀DEBUG ␛[0m| ??:??:?? 4 ␛[34m␀Total PSRAM: 0
␛[0m␀␛[34m␀DEBUG ␛[0m| ??:??:?? 4 ␛[34m␀Free PSRAM: 0

...

␛[0m␀␛[32m␀INFO  ␛[0m| ??:??:?? 6 ␛[32m␀Init Store & Forward Module in Server mode
␛[0m␀␛[32m␀INFO  ␛[0m| ??:??:?? 6 ␛[32m␀S&F: device doesn't have PSRAM, Disable
```

JohnTaipei, 2025-01-23:
T190
no psram

JQ3B94, 2025-02-13:
both boards have ESP32-S3R8 MCU's ( 8MB of PSRAM) so this is in fact an actual bug 
the hardware should support the feature

closed 2025-04-07 by thebentern