meshtastic/firmware #5911: [Feature Request]: Basic send/receive messages via access point on ESP32

redfast00, 2025-01-22:
### Platform

ESP32

### Description

Is there any work on making an ESP32 meshtastic node start an AP, host a web server and small web UI to send/receive messages? The advantage over having to use bluetooth and the meshtastic app, is that you don't need the meshtastic app, so you can hand out nodes to people and they can use them from their phone.

Someone told me that this used to be supported, but was cut in the 2.0 release because it didn't work very well. What is the main blocker to implement this?

madeofstown, 2025-01-22:
Get an Android device. ESP32 devices support WiFi already, it's the the Apple App that does not support connecting with IP address.

fifieldt, 2025-01-22:
This is an existing feature.

fifieldt, 2025-01-22:
Ah, sorry, early morning - I misread.

caveman99, 2025-01-22:
> Someone told me that this used to be supported, but was cut in the 2.0 release because it didn't work very well. What is the main blocker to implement this?

The Node runs really hot in AP mode which makes it unstable. Also people were having expectations it would never be able to meet. AP Mode in ESP32 is essentially a way to configure it, while people were hpoping for some kind of hub to share the one radio connection.

garthvh, 2025-01-24:
The he issue is not power consumption, the captive portal and bad user experience of being connected to an ap with no internet is the issue.

redfast00, 2025-01-25:
@garthvh how I envision this, is that it's not for regular users to use, they would still use the app + Bluetooth; but in emergency situations where there is no internet, you can hand out nodes to people without having them have to install a new app (which they can't do since there is no internet).

redfast00, 2025-01-25:
The web interface could even say 'This is not the best way to use Meshtastic, for a better interface, please use the app' or something

fifieldt, 2025-01-25:
@redfast00 what about pre-configuring the nodes to connect to a specific SSID, then setting up a hotspot on the phone with that SSID?

redfast00, 2025-01-25:
@fifieldt that's a useful alternative, but it's a lot harder to do for non-tech savvy users: instead of just tapping a network to connect and entering the password (an action they do regularly), they have to set up an access point, configure it right, somehow find the AP of the mesh node, enter that in their browser (possibly also with a port number, but I guess we could make this :80). On my phone (which is a pretty recent Android phone), setting up a hotspot does not work.

I'm looking for something I can just hand out with minimal instructions or handholding needed.

garthvh, 2025-01-25:
Esp32 ap mode is all handholding, is a really bad user experience with modern phones, it won't work as you are expecting.

fifieldt, 2025-01-25:
@redfast00 had a look at esp32-open-mac, nice work! Based on your experience, would you be able to estimate how much extra resource would be required on an esp32 device to enable AP mode? Our firmware is packed in pretty tight.

fifieldt, 2025-01-25:
old softAP threadhttps://github.com/meshtastic/firmware/issues/1835

closed 2025-01-25 by fifieldt

redfast00, 2025-01-25:
> [@redfast00](https://github.com/redfast00) had a look at esp32-open-mac, nice work! Based on your experience, would you be able to estimate how much extra resource would be required on an esp32 device to enable AP mode? Our firmware is packed in pretty tight.

Sure :)  As far as I know, the main resource use with SoftAP mode is not CPU/RAM, but the amount of packets it sends: in the default configuration, it will send a beacon packet every 100ms, so 10 times per second where it uses the 2.4GHz radio to transmit. I haven't tested this (for lack of a power profiler) but I suspect that reducing the beacon interval will reduce the power consumption by a lot. I'd also have to check what hardware filter configuration they use: the hardware supports filtering packets based on MAC address to reduce the amount of packets they have to handle in software.

fifieldt, 2025-01-25:
I'm guessing you'd also need a dhcp server and maybe a DNS server, as well as captive portal software to redirect the http requests to the dashboard?