meshtastic/firmware #5913: How read information from node in C?

lykamspam, 2025-01-22:
### Platform

ESP32

### Description

I need information about nodes.
Similar python "meshtastic --nodes"

How I can to do this? Here is my code. What is wrong? I'm not get any data. I use Heltec 3

```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <fcntl.h>
#include <unistd.h>
#include <termios.h>
#include <errno.h>

#define SERIAL_PORT "/dev/ttyUSB0" 
#define BAUDRATE B115200  

// Funkcja do wysyłania i odbierania danych
void send_and_receive(int serial_port, const char *command) {
    printf("Wysyłana komenda: %s\n", command); // Debugowanie
    write(serial_port, command, strlen(command));

    // Czekaj na odpowiedź
    usleep(2000000);  
    char response[256];
    int num_bytes = read(serial_port, response, sizeof(response) - 1);
    if (num_bytes > 0) {
        response[num_bytes] = '\0'; // Zakończenie łańcucha
        printf("Otrzymano odpowiedź: %s\n", response); // Przekazanie odpowiedzi
    } else if (num_bytes < 0) {
        perror("Błąd podczas odbierania danych");
    } else {
        printf("Brak odpowiedzi.\n");
    }
}

int main() {
    int serial_port;
    struct termios options;

    // Otwieranie portu szeregowego
    serial_port = open(SERIAL_PORT, O_RDWR | O_NOCTTY | O_NDELAY);
    if (serial_port < 0) {
        perror("Nie można otworzyć portu szeregowego");
        return -1;
    }

    // Ustawienia portu szeregowego
    tcgetattr(serial_port, &options);
    cfsetispeed(&options, BAUDRATE);
    cfsetospeed(&options, BAUDRATE);
    options.c_cflag |= (CLOCAL | CREAD); // Ustawienia lokalne i odczyt
    options.c_cflag &= ~PARENB; // Brak parzystości
    options.c_cflag &= ~CSTOPB; // 1 bit zatrzymania
    options.c_cflag &= ~CSIZE;
    options.c_cflag |= CS8; // 8 bitów danych
    options.c_iflag &= ~(IXON | IXOFF | IXANY); // Wyłączenie kontroli przepływu
    options.c_lflag &= ~(ICANON | ECHO | ECHOE | ISIG); // Wyłączenie trybu kanonicznego i echo
    options.c_cc[VMIN] = 0;  // Minimalna liczba znaków do odbioru
    options.c_cc[VTIME] = 10; // Czas oczekiwania w dziesiątych częściach sekundy
    tcsetattr(serial_port, TCSANOW, &options);

    // Wysyłanie polecenia do uzyskania węzłów
    const char *command = "{\"cmd\": \"get\", \"args\": {\"type\": \"nodes\"}}\n";
    send_and_receive(serial_port, command);

    // Zamknięcie portu szeregowego
    close(serial_port);
return 0;
}
```

esev, 2025-01-22:
Hi @lykamspam 

This guide has more information about how to use the serial interface: https://meshtastic.org/docs/development/device/client-api/

GUVWAF, 2025-01-22:
The Arduino client might also be of help: https://github.com/meshtastic/Meshtastic-arduino

i0vt, 2025-01-23:
Both links do not contain any code example. I too would like to write a Linux terminal client. Maybe just correct the example at the top. Throwing links that don't give anything is pointless.

GUVWAF, 2025-01-23:
The Arduino client is an example of how to implement an interface in C++: https://github.com/meshtastic/Meshtastic-arduino/blob/master/src/mt_protocol.cpp 

Yes, it’s not C, but I don’t know of any C implementation.