meshtastic/firmware #5915: [Bug]:

naskobogdanov, 2025-01-23:
### Category

Other

### Hardware

Heltec Wireless Paper

### Firmware Version

2.5.11.8e2a3e5 Beta +

### Description

GPS on Wireless Paper is not recognized on 2.5.11 onwards. The last version it was working properly is 2.5.9. GPS_RX_PIN = 37

### Relevant log output

```Shell

```

thebentern, 2025-01-23:
The Heltec Wireless Paper does not have a GPS module nor does it have any exposed GPIO to use an external one.

closed 2025-01-23 by thebentern

naskobogdanov, 2025-01-23:
As I mentioned here:
https://meshtastic.discourse.group/t/adding-gps-to-heltec-wireless-paper/12882/3

It is quite easy to add GPS to Wireless Paper. Can I use other firmware so I can have the latest firmware with GPS functionality or is there instructions how to compile it myself?

naskobogdanov, 2025-01-24:
@fifieldt Thank you for the tip. 
Compiling is not a problem. Checking the varinat.h file for the wireless paper made me think that I just need to change NO_EXT_GPIO to 0, but it didn't work. This problem was persistent even before implementing the NO_EXT_GPIO change. 
So following the logs I got this change in the GPS.cpp between 2.5.10 and 2.5.11 that gets the GPS not to work for me:

```
475    if (gnssModel != GNSS_MODEL_UNKNOWN) {
476               setConnected();
477           } else {
478               return false;
479          }
```

I've set it to "true" in order to work with my cheap Chinese GPS, because I do now know how to handle it in the variant.h
Possibly this could affect other devices with random GPS modules. 
Thanks again for the heads up.

fifieldt, 2025-01-24:
Happy to help. If you can send in logs during the early phase of the boot when it's doing GPS detection in another bug report I can take a look at adding support for your GPS