meshtastic/firmware #5933: [Bug]: ina3221 inverts current over certain values (maybe signed int overflow?)

sgtwilko, 2025-01-25:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.4.* to at least 2.5.15

### Description

I have an ina3221 setup with input 1 showing power being used, input 2 showing solar being provided, and input 3 showing the actual current flowing in out of the battery.

The first two should always be positive, the third should be positive when drawing current from the battery and negative when it's being charged.

With low current this is always the case, when one of them goes over about 250mA it inverts that value.

I suspect this is a mismatch in int lengths between the library that's taking the values and where it's being stored as it's typical of a 2s compliment overflow (although strangely would seem to be 9 bits)

### Relevant log output

```Shell

```

b8b8, 2025-01-25:
Have you correctly changed the address of the INA3221 off the default so it is correctly recognized by the firmware?
You need to bridge SDA.

[image]

isseysandei, 2025-01-25:
You beat me to it, I was going to open a report about this soon... Same issue as you, but after a while it fixes by itself, idk if happen the same to you.

[image]

sgtwilko, 2025-01-26:
> Have you correctly changed the address of the INA3221 off the default so it is correctly recognized by the firmware?
> You need to bridge SDA.
> 

Yes, you don't get any values at all if you haven't changed the address.

You can see from the images here in getting values, and that they suddenly reverse ( particularly channel 2, the solar input)

[image]

[image]

[image]

This does seem very typical of a twos compliment overflow. Where large values cause the first bit of the word to become a 1 which is then interpreted as a negative value.

sgtwilko, 2025-01-26:
> You beat me to it, I was going to open a report about this soon... Same issue as you, but after a while it fixes by itself, idk if happen the same to you.
> 

From my limited observations, once it falls back below a threshold it reverts back to reporting correctly.

That seems slightly different to what we see in your graph.

dblmca, 2025-01-30:
I sent the dev a note about this back in Nov.
I don't have the code in front of me, but the issue was something like shunt_uV becomes negative due to overflow the calculated current becomes incorrect.

The fix should be to use an int32_t in getshuntvolt

fifieldt, 2025-01-30:
Had a quick look in the code and there is indeed a uint16 in there at the moment.

sgtwilko, 2025-01-30:
> I sent the dev a note about this back in Nov.
> I don't have the code in front of me, but the issue was something like shunt_uV becomes negative due to overflow the calculated current becomes incorrect.
> 
> The fix should be to use an int32_t in getshuntvolt 

I also had a look, before seeing your comment.

It's absolutely in need of an int32_t as the multiplication by 40 overflows the int16.

Going to try this on my local and see if it works

sgtwilko, 2025-02-06:
I've now managed to confirm the figure using a multimeter to monitor the current in line with one of the channels.

It is correctly reporting current up to an amp now.

I'll get the PR opened today.

dblmca, 2025-02-06:
> I've now managed to confirm the figure using a multimeter to monitor the current in line with one of the channels.
> 
> It is correctly reporting current up to an amp now.
> 
> I'll get the PR opened today.

Thank you for staying on top of this, I've been meaning to check how I did it back in Nov, but... life. Will be nice not to have to worry about one more library during builds.
Cheers!

dblmca, 2025-02-07:
has anyone been able to check if the INA3221 still works in .19, .20 or .21?
I just tried and it didn't seem to work, but I'm running wonky custom hardware, and it could just be me.

sgtwilko, 2025-02-07:
I'm running a custom built .20,
It works for power telemetry, but I've been having issues getting it to be used for battery %.

isseysandei, 2025-02-07:
> has anyone been able to check if the INA3221 still works in .19, .20 or .21? I just tried and it didn't seem to work, but I'm running wonky custom hardware, and it could just be me.

I flashed .21 on my NRF nodes and I noticed that Battery % from either INA3221 and 219 is broken. Also, idk if it related, but the INA219 values are no longer passed by ch3 curr/voltage. It stops broadcasting that packet, no problem with the INA3221...

sgtwilko, 2025-02-07:
Sounds like a different issue to this one, but worth comparing the source from an earlier version to .20
When did it work last?

dblmca, 2025-02-07:
> Sounds like a different issue to this one, but worth comparing the source from an earlier version to .20 When did it work last?

I went back to the .18 code base and its working fine.
Ill take a look at the later ones this weekend to see what's up.

lagunacomputer, 2025-02-08:
i use .18 and had to switch android app option Power Config -> Battery INA_2XX I2C address value to 66 to get it to report channel 1 as battery, and channel 2 as solar power values in the right order.  I may have also edited the i2c address in code but i can't find the line or file it was in

dblmca, 2025-02-28:
INA3221 current reading seems to be broken on 2.6.0 anyone else get the chance to try it out?

tache, 2025-04-07:
Hello all ...

Using .20, I am seeing something like this. Setup is a INA3221 via the MCSBC-SVR module. Power telemetry packet is reporting stale measurements (pre-integration INA3221 values). The INA telemetry packet is correct, confirmed with a MM. You can see they were sent quickly in sequence.

Power
```json
{
  "id": 3853892436,
  "payload": {
    "air_util_tx": 0.317694425582886,
    "battery_level": 18,
    "channel_utilization": 4.54333353042603,
    "uptime_seconds": 22124,
    "voltage": 3.3989999294281
  },
  "timestamp": 1743991343,
  "type": "telemetry"
}

```

INA3221 - inside, not solar.
```json
{
  "id": 3545676627,
  "payload": {
    "current_ch1": 1.20000004768372,
    "current_ch2": 3.59999990463257,
    "current_ch3": 0,
    "voltage_ch1": 4.15999984741211,
    "voltage_ch2": 3.92000007629395,
    "voltage_ch3": 0
  },
  "timestamp": 1743991313,
  "type": "telemetry"
}
```

MOE303, 2025-05-15:
I also use the INA3221 to measure currents in my solar node and unfortunately found that even in firmware 2.6.4 the issue with changing values from positiv to negative (or vice versa) still occurs.

As soon as a value exceeds a magic limit of approx. 326mA the the value changes to -326mA and rises back towards 0. (336mA will be shown as -316mA and so on...)
This is very confusing if you want to check if the battery is charging or discharging.

Is there a chance to solve this issue in one of the next firmware versions? 

All the best and thanks for your great work!

dmarman, 2025-05-27:
Should we fork the current INA3221, merge the PR from @sgtwilko and update the dependency on the platform.ini? It looks like the maintainer of the INA3221 is not responding for months.

Another option is to use Adafruit library (most of the other sensors use Adafruit), which uses floats for the measurements.

I am happy to work in any of those solutions. I am of course facing this issue too with my solar node.

closed 2025-09-04 by fifieldt