meshtastic/firmware #5935: [Bug]: No focus on new messages PR #5881broke a very important feature for Standalone nodes

HarukiToreda, 2025-01-25:
### Category

Other

### Hardware

Heltec Wireless Paper

### Firmware Version

2.5.20

### Description

One of the most critical features of Meshtastic is its messaging functionality, especially for standalone units. Removing message priority from the node, as seen in PR #5881, undermines the utility of these devices entirely. When a message is received, users are now left guessing if anything new has arrived, as there is no longer a clear way to identify new messages. This significantly impacts the usability and reliability of standalone units for the broader community.

The use case of that PR seems very niche unique to this one user vs the community that uses standalone units. In this video you can see there is no way for me to know I received a message. I hope this can be reverted and a better approach can be found.

https://github.com/user-attachments/assets/cead01a8-c331-41c9-ac32-36ca043bac35

### Relevant log output

```Shell

```

isseysandei, 2025-01-25:
The part I added should take action only when the caurosel is set to zero, have you tried enabling it, or setting a very high screen change time?

HarukiToreda, 2025-01-25:
> The part I added should take action only when the caurosel is set to zero, have you tried enabling it, or setting a very high screen change time (if you don't need the screen to change but you want to focus on incoming messages)?

standalones do not use the carousal feature, that is used for people who want to see all screens and rarely used for standalones that are messaging centric.

isseysandei, 2025-01-25:
Fair points, my end goal was creating something like a toggable option to disable focuses, but I have zero experience in Kotlin or the IOS equivalent...
As temporary workaround we can set a specific value (idk 1337?) instead of activating it at zero.

isseysandei, 2025-01-25:
> what exactly are you monitoring on the node that you can't monitor on the app?

One example:
In a node set up as MQTT gateway or simply if you want to use it from wifi: The screen showing the ip address, expecially usefull if you are unable to set a static address and it changes after a reboot/IP reservation expiration.

HarukiToreda, 2025-01-25:
the proper approach would need to add a setting on the display settings disabling message focusing, but that requires a more broad approach

isseysandei, 2025-01-25:
I'm working on a better implementation now

kalestew, 2025-01-29:
Yeah, I can see how this change might be frustrating on the tdeck (been using the TFT ui for so long now I forgot about that). 

This change is long overdue though; at the very least an option to disable this behavior is desperately needed. The auto-focus renders the screen useless for the few times it might be handy to the majority of users who are connecting these to an app. 

The workaround of setting the carousel interval very high is a good middle-ground to achieve the old behavior but a separate configuration toggle would be the best option -- let the users have choice!

HarukiToreda, 2025-01-31:
Since the issue has been solved by being pulled back from the main branch, I am closing this PR. When a better solution is made, I'm sure it will be much better.

closed 2025-01-31 by HarukiToreda