meshtastic/firmware #5937: [Bug]: GPS Time is *NOT* well synchronized from devices.

jstockdale, 2025-01-25:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.19

### Description

Hi folks,

I'm watching my GPS time on my SenseCap Indicator which is connected to a Air530 GPS from SeeedStudio. This is all supported hardware.

The GPS time does *NOT* regularly sync. Maybe once a day or something? It is constantly off and is almost never correct. Watching the serial output from the GPS unit, the time is exactly correct (maybe ~250ms off from the serial delay).

If GPS time is available and the device has a good fix. The device should *prefer* GPS time over mesh time. Or use NTP if NTP is set...

It seems to do none of these things. I'm guessing it's syncronizing like once every 24 hours. Either we should syncronize GPS time every time position is updated, or we should have a way to prefer NTP time.

Having the wring time continually on a GPS connected device is suboptimal.

Thanks,
-@jstockdale

### Relevant log output

```Shell

```

fifieldt, 2025-01-25:
Thanks for sending in a report. I have that hardware and will try to reproduce.

fifieldt, 2025-02-08:
Hi @jstockdale , I set up the Sensecap Indicator with an Air530 attached ... but the GPS was never actually detected.

It looks like the Grove connectors on the back connect to the RP2040, whereas the Meshtastic firmware runs on on the RP2040.

Further, I looked into the time code, and it looks like it does function as one would expect - applying GPS time in favour of time from mesh and time from NTP.

How have you got the GPS connected? Can you provide logs (https://flasher.meshtastic.org --> open serial monitor)

mverch67, 2025-02-23:
To get the GPS working with indicator you'd need to flash this firmware to the rp2040 as documented in PR #5411:
https://github.com/Dylanliacc/Serial-Protocol-of-RP2040-on-Indiactor-for-meshtastic

closed 2025-02-23 by fifieldt