meshtastic/firmware #5940: [Feature Request]: Add Regions ZA_868 (1% Duty Cycle) & ZA_433  (10% Duty Cycle)

martinville, 2025-01-26:
### Platform

other

### Description

Many users have the misconception that its safe to use EU_433 or EU_868 in South Africa. Also some users also believe that a duty cycle of 10% is OK on Freq 868 but its not the case.

In South Africa according to the local ICASA regulation the legal settings are as per the below (Same as existing Ukraine region profile UA_868 & UA_433)

433Mhz 10% duty cycle
868Mhz 1% duty cycle

Is this something you are willing to add ? Then at least you also have one setting for Africa :-)

Source:
https://www.icasa.org.za/uploads/files/Notice-of-intention-to-amend-Annexure-B-of-the-Radio-Frequency-Spectrum-Regulations.pdf

GUVWAF, 2025-01-26:
`EU_868` uses the 250kHz portion from 869.4-869.65 MHz, which is listed as 10% duty cycle at 500mW (27dBm) in the document you referenced: 

[image]

martinville, 2025-01-26:
Thanks for confirming, further up previous page it shows the below though .

[image]

GUVWAF, 2025-01-26:
It's called EU_868 since the band is often referred to as the 868MHz band, but it uses specifically 869.4-869.65MHz: 
https://github.com/meshtastic/firmware/blob/10d553087c4d8158b961a5b9ab8b6a1ff413ca38/src/mesh/RadioInterface.cpp#L47

martinville, 2025-01-26:
Thanks for this, also good news!

fifieldt, 2025-01-27:
To summarise:
ZA_868 == EU_868
ZA_433 == UA_433

Generally, we don't add 'duplicate' regions into the firmware. However, it might be worth asking the apps whether they can apply a nicer treatment as they have country names in there.

GUVWAF, 2025-02-15:
Please open issues for the relevant apps if you want to clarify which region should be used for South Africa, and maybe consider adding it here: https://meshtastic.org/docs/configuration/region-by-country/ (use the "Edit this page" button).

closed 2025-02-15 by GUVWAF