meshtastic/firmware #5944: [Feature Request]: Ability to customize where linux hosts load the TLS key material from

kquinsland, 2025-01-27:
### Platform

Linux Native

### Description

I am in the process of adding LetsEncrypt/Certbot to my meshtastic-on-debian node and was curious about where the key material lived so that I could provide my own...

-----

After a bit of digging, found that the web server will [generate it's own certificate if one can't be found](https://github.com/meshtastic/firmware/blob/4747e73f37e8f6d14aaa3288841fa0f7d8f1c177/src/mesh/raspihttp/PiWebServer.cpp#L450-L455)

But where does the webserver look for the certificate?
I can't find any directive in the [`config.yaml` file](https://github.com/meshtastic/firmware/blob/4747e73f37e8f6d14aaa3288841fa0f7d8f1c177/src/platform/portduino/PortduinoGlue.cpp#L150C21-L150C38) that would let me set this :(.

I can see that the key material has [hard-coded names](https://github.com/meshtastic/firmware/blob/4747e73f37e8f6d14aaa3288841fa0f7d8f1c177/src/mesh/raspihttp/PiWebServer.cpp#L387): `certificate.pem` and `private_key.pem`.

But those are just file _names_... there's no path/folder structure in the code!

Where do those files live? Let's try to `find` them: 

```shell
root@meshtastic:/# find / -name "private_key.pem"
/private_key.pem
```

One result and it's in the **root**!?!! I bet it's sibling is there, too:

```shell
root@meshtastic:/# ls -lah *.pem
-rw-r--r-- 1 root root 1.1K Jan 18 19:02 certificate.pem
-rw-r--r-- 1 root root 1.7K Jan 18 19:02 private_key.pem
```

The file names match up, but let's confirm:

```shell
root@meshtastic:/# openssl x509 -in /certificate.pem -noout -fingerprint -sha256
sha256 Fingerprint=56:D3:F6:<omitted>:0B:CD:30
root@meshtastic:/# openssl s_client -connect meshtastic.local:443 -servername meshtastic.local < /dev/null | openssl x509 -noout -fingerprint -sha256
depth=0 C = DE, O = Meshtastic, CN = meshtastic.local
verify error:num=18:self-signed certificate
verify return:1
depth=0 C = DE, O = Meshtastic, CN = meshtastic.local
verify return:1
DONE
sha256 Fingerprint=56:D3:F6:<omitted>:0B:CD:30
```

Yep! The fingerprint for the file (`/certificate.pem`) matches the fingerprint from the web server.

-----

# The Ask

Please extend the [`config.yaml` file](https://github.com/meshtastic/firmware/blob/4747e73f37e8f6d14aaa3288841fa0f7d8f1c177/bin/config-dist.yaml#L182C1-L184C64) to allow setting a directory where key material can be set/read from.
The "generate if not found" behavior is good and should be kept, the ask is for a user-configurable directory to be used instead of `/`.

Ideally something like this:

```yaml
Webserver:
  Port: 443 
  RootPath: /usr/share/meshtasticd/web
  tls:
    # Dir that is prefixed to argument fed to `read_file_into_string()` calls
    path: /etc/meshtasticd/pki       
    # Allow specifying the cert/key name. If not explicitly set, keep the defaults
    cert: certificate.pem                  
    key: private_key.pem
```

Which would result in the webserver reading key material from `/etc/meshtasticd/pki/{certificate,private_key}.pem`

## The why

There are a few reasons but they boil down to

- Nobody likes a messy `/`
- I am trying to build a more sophisticated node that leverages a read-only file system for _most_ things. It is impossible to do this well if `/` needs to be writable. Ideally, only the `/root/.portduino` directory would be on a writable partition and everything else would be loaded from a read only partition
- It's a lot easier/cleaner if I can pick the directory where the file(s) are read from, especially for hooks/scripts that are called after CertBot gets a certificate issued to it.

-----

Related, I think but I'm not sure if the ticket is just asking for TLS support or the ability to change settings about TLS.

https://github.com/meshtastic/firmware/issues/5740

fifieldt, 2025-01-27:
Thanks for the detailed investigation. Yes, specifying TLS cert location is a fairly standard ask for linux daemons, so we should probably do this :)

closed 2025-02-01 by fifieldt

vidplace7, 2025-02-01:
@kquinsland thanks again for suggesting this.

Please give the new functionality a test when you get a chance 🙂 Should be included in tonight's "daily" builds.

https://launchpad.net/~meshtastic/+archive/ubuntu/daily
https://build.opensuse.org/package/show/network:Meshtastic:daily/meshtasticd
https://copr.fedorainfracloud.org/coprs/g/meshtastic/daily/

kquinsland, 2025-02-01:
> Please give the new functionality a test when you get a chance 🙂 Should be included in tonight's "daily" builds.

Awesome, i'll look.

If I'm reading GHA, right, [this](https://github.com/meshtastic/firmware/actions/runs/13087136295/job/36519778229) GHA run was done on `master@4c0e0b84712e08371d84400187e42996bb5bc254`  but I'm not finding raspberry pi os nightly debs.

The dependencies for ubuntu are all _slightly_ too new for rPi OS :(.

vidplace7, 2025-02-02:
> Awesome, i'll look.
>
> If I'm reading GHA, right, [this](https://github.com/meshtastic/firmware/actions/runs/13087136295/job/36519778229) GHA run was done on [...]

https://software.opensuse.org//download.html?project=network%3AMeshtastic%3Adaily&package=meshtasticd

Use the guide here, select "Raspbian" and you should be off to the races 👍

kquinsland, 2025-02-02:
> https://software.opensuse.org//download.html?project=network%3AMeshtastic%3Adaily&package=meshtasticd
> 
> Use the guide here, select "Raspbian" and you should be off to the races 👍

TIL, suse infra building deb files :D.

-----

```shell
karl@meshtastic:/tmp $ sudo systemctl stop meshtasticd
karl@meshtastic:/tmp $ wget https://download.opensuse.org/repositories/network:/Meshtastic:/daily/Raspbian_12/arm64/meshtasticd_2.5.21.34~obs4c0e0b8~unstable_arm64.deb
karl@meshtastic:/tmp $ sudo dpkg -i meshtasticd_2.5.21.34~obs4c0e0b8~unstable_arm64.deb 
<...>
karl@meshtastic:/tmp $ sudo systemctl start meshtasticd
<...>
```

Seems to have done the trick. Key material permissions remain the same but the new location is respected.

PR #5961 (2025-02-01): Portduino: Set Web SSL Cert / Key paths from yaml
Allows SSL web certificate / key locations to be set in the meshtasticd config.yaml.

Additionally, sets the default directory for ssl storage to `/etc/meshtasticd/ssl`, and creates this directory in the Linux packaging.

Example:
```yaml
Webserver:
  Port: 443 # Port for Webserver & Webservices
  RootPath: /usr/share/meshtasticd/web # Root Dir of WebServer
  SSLKey: /etc/meshtasticd/ssl/private_key.pem
  SSLCert: /etc/meshtasticd/ssl/certificate.pem
```

Resolves #5944