meshtastic/firmware #5946: [Bug]:

ve7wdm, 2025-01-27:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.5.18.89

### Description

LilyGo T-Echo will display correct GPS coordinates on its own display, however GPS coordinates displayed within the Meshtastic app (Android) are not the same coordinates as on the T-Echo resulting in map locations being off by several kilometers.  This is true with three T-Echo devices using firmware 2.5.18.89 and app version 2.5.16 and all three units exhibit same behaviour.  Attempted multiple reflashings of device (also clearing of flash memory) to no avail.  Attempted to install app on clean Android device (GrapheneOS) to no avail.

### Relevant log output

```Shell

```

garthvh, 2025-01-27:
Check the position precision on your channel, the default is not to leak your personal I fo

closed 2025-01-27 by garthvh

fifieldt, 2025-01-27:
Thanks for using meshtastic!

This is expected behaviour - we try to protect your privacy by default. Have a look at the position precision settings to change it :)

https://meshtastic.org/docs/configuration/radio/channels/#position-precision

ve7wdm, 2025-01-27:
Oh my goodness, thank you so much. I don't know how I missed that setting.  :-)

---- On Sun, 26 Jan 2025 18:39:01 -0800 Tom Fifield ***@***.***> wrote ---

Thanks for using meshtastic!

This is expected behaviour - we try to protect your privacy by default. Have a look at the position precision settings to change it :)

https://meshtastic.org/docs/configuration/radio/channels/#position-precision