meshtastic/firmware #5949: [Feature Request]: Make meshtastic devices discoverable by the Apple FindMy network

glitch003, 2025-01-27:
### Platform

NRF52, ESP32, RP2040

### Description

Reverse engineering of the FindMy network has revealed that there is no authentication needed to advertise location by a bluetooth beacon.  Therefore, it's possible to make Meshtastic devices show up on FindMy if the user has properly downloaded a private key from the device.  This would be a nice bonus feature for Meshtastic devices to have, in the case where they are out of range of other Meshtastic devices, but within range of any iPhones.

More info is here: https://github.com/seemoo-lab/openhaystack

Note: from the firmware PoV, no integration with Apple services, or an Apple account, is needed.  All that is needed is for the beacon protocol to conform.  To actually track the device on the FindMy network, you do need an apple account.  In practice, this is useful when a user has noticed that a device has not reported it's location on the Meshtastic app for some time, and they could switch over to the FindMy app as a backup to see it's location.

be-a-panther, 2025-04-01:
Maybe this projects helps to implement the google spinoff: https://github.com/leonboe1/GoogleFindMyTools

sendingstrings, 2025-09-09:
Maybe a question is, what current Meshtastic devices would be capable of this?
What is required on the BT HW side to support this? 
What is needed to support both this and allow for a BT client connection?

EON23, 2025-09-17:
In terms of what hardware and firmware is needed to make this work, as long as the device can broadcast an advertisement pocket over Bluetooth at a set interval, this should work. You dont need an actual connection to a device, its a bit like the meshtastic lora broadcasts in that sense. Both the NRF and ESP are capable of this, I believe the airtags actually use an NRF chip as well. 
The issue is generating that token for Google or Apple devices is a little convoluted. Ive tried the google github project for diy trackers with an esp and it worked but theres a couple issues such as location not being accessible in in the google app. It requires a python script to manually grab it from googles servers.
I believe theres is a similar issue with the apple side of things as well although I havent tried it personally, where you need to use their app to grab the diy airtag locations.
This could be resolved by integrating the location fetching from google and apple servers it into the meshtastic app but then your technically misusing their infrastructure which could lead to some potential legal trouble.
That being said its technically possible with the current hardware, maybe an arbitrary bluetooth broadcast feature could be implemented then its up to the user how they deal with handling the google/apple side of things?

ciavolella, 2025-09-29:
This is a project I plan to expend some effort on.  I want a good way to track my kids, and have a way to tell them it's time to come home without giving them cell phones.  GPS is no-go, because if Lora is out then I can't query location.  Anyway, I'm just happy I'm not the only one who sees the utility in this, and am glad I (hopefully) wont be the only one working on pairing the two.

garthvh, 2026-09-03:
There is additional certification required to show up on either apple or google's tracking system,  This would mostly alert users that their meshtastic radio is stalking them, not a real good user experience available currently

closed 2026-09-03 by garthvh