meshtastic/firmware #5960: [Bug]: Wio-E5 doesn't accept any configuration settings

dchokola, 2025-01-29:
### Category

Other

### Hardware

Other

### Firmware Version

2.5.20.4c97351

### Description

Tried with 2.5.20.4c97351 and 2.5.20.4c97351. I completely erased the flash and programmed the .elf.
Setting any configuration value at all does not result in a change in configuration on the board. For example:
```
dan@ender:~/.local/lib/python3.13/bin$ ./meshtastic --port /dev/ttyUSB0 --set lora.region US
Connected to radio
Set lora.region to US
Writing modified preferences to device

dan@ender:~/.local/lib/python3.13/bin$ ./meshtastic --port /dev/ttyUSB0 --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
```

I've also tried importing a .yaml file from another radio, but when reading back the configuration with --info, all fields are at their default setting.

### Relevant log output

```Shell

```

caveman99, 2025-01-29:
The STM32 platform does not have a filesystem or room for one. You need to burn the settings into the elf during compilation with the userprefs.jsonc file.

closed 2025-01-29 by caveman99

dchokola, 2025-01-29:
Sorry about the non-issue. Is that documented anywhere? I found very little about the STM32 variants in the official documentation and nothing in the released .elf or .bin zip files, platformio.ini, or variant.h.

markbirss, 2025-01-29:
> Sorry about the non-issue. Is that documented anywhere? I found very little about the STM32 variants in the official documentation and nothing in the released .elf or .bin zip files, platformio.ini, or variant.h.

modify
https://github.com/meshtastic/firmware/blob/master/userPrefs.jsonc

"USERPREFS_CONFIG_LORA_REGION": "meshtastic_Config_LoRaConfig_RegionCode_US",

in your local build directory and compile