meshtastic/firmware #5969: [Bug]: INA226 not reporting correctly

morcant, 2025-01-31:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.18

### Description

I have this INA226 module connected via i2c bus to wisblock: https://www.amazon.com/INA226-Voltage-Current-Module-Monitoring/dp/B08MKSQSL9

I have it wired up like this, though I have tried a variety of other wiring methods as well:
[image]

The voltage is off by a factor of 1000. (0.0135 vs 13.5V) and current is not reported at all, when I know there is 0.05A of current.

I have this same model in use on an esp32 with esphome wired this way, and it's reporting voltage and amperage correctly. I cannot easily swap them to see if this unit is defective, but my suspicion is that since support was just added in recent firmware that it may be a bug in meshtastic code.

### Relevant log output

```Shell
INFO | 19:46:54 19 [PowerTelemetry] Power Telemetry: init
INFO | 19:46:54 19 [PowerTelemetry] Init sensor: INA226
INFO | 19:46:54 19 [PowerTelemetry] Opened INA226 sensor on i2c bus

INFO | 20:09:55 1400 [PowerTelemetry] Send: ch1_voltage=0.000000, ch1_current=0.000000, ch2_voltage=0.000000, ch2_current=0.000000, ch3_voltage=0.013459, ch3_current=0.000000
DEBUG | 20:09:55 1400 [PowerTelemetry] Partially randomized packet id 2217119550
INFO | 20:09:55 1400 [PowerTelemetry] Send packet to phone
```

fifieldt, 2025-02-01:
Thanks for the report! This is a newly-supported sensor, so this may well be a bug. Having a look at the code.

fifieldt, 2025-02-01:
Yup - it looks like inside meshtastic we use milliVolts but the sensor library returns volts. Adding code to multiply that by 1000.

The lack of current is not so straightforward - everything looks to be in milliAmps there.

morcant, 2025-02-03:
Confirmed voltage reporting correctly now. I'll look into amperage a bit more and open a new bug if I identify anything.

closed 2025-02-03 by thebentern