meshtastic/firmware #5970: [Bug]:

Sabering1, 2025-01-31:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.18.89ebafc Beta

### Description

I installed latest firmware on seedstudio 
XIAO ESP32S3 & Wio-SX1262 Kit for Meshtastic & LoRa
However i get a error when i want to send a message saying: "No interface" on the android app, also it wont load setting normaly and takes like 2 minutes to load and save them

### Relevant log output

```Shell

```

DrMecq, 2025-02-07:
Hi,

I had the same problem but everything changed when I connected pins according to:

https://adrelien.com/blog/meshtastic-diy-how-to-build-your-own-meshtatic-node-esp32-lora-radio/

Try to connect them with the same way.

Sabering1, 2025-02-07:
https://www.seeedstudio.com/Wio-SX1262-with-XIAO-ESP32S3-p-5982.html?qid=eyJjX3NlYXJjaF9xdWVyeSI6Im1lc2h0YXN0aWMiLCJjX3NlYXJjaF9yZXN1bHRfcG9zIjoyLCJjX3RvdGFsX3Jlc3VsdHMiOjEwLCJjX3NlYXJjaF9yZXN1bHRfdHlwZSI6IlByb2R1Y3QiLCJjX3NlYXJjaF9maWx0ZXJzIjoic3RvcmVDb2RlOltyZXRhaWxlcl0ifQ%3D%3D

Im talking about this, there are no pins to connect, its a all in one solution

caveman99, 2025-02-07:
Working fine here, have you tried a factory restore?

Sabering1, 2025-02-07:
Yes, but i still get the same error

caveman99, 2025-02-07:
We need the serial debug log then.

fifieldt, 2025-02-07:
@Sabering1 go to https://flasher.meshtastic.org and tap on open serial monitor.

Sabering1, 2025-02-13:
[meshtastic-log-2025-02-13T22-47-19.235Z.log](https://github.com/user-attachments/files/18790992/meshtastic-log-2025-02-13T22-47-19.235Z.log)

GUVWAF, 2025-02-15:
Please double-check that you flashed the firmware for the correct device.

fifieldt, 2025-02-16:
> [meshtastic-log-2025-02-13T22-47-19.235Z.log](https://github.com/user-attachments/files/18790992/meshtastic-log-2025-02-13T22-47-19.235Z.log)

Thanks for the logs. Could you capture them just after a reboot, too?

Sabering1, 2025-02-16:
> > [meshtastic-log-2025-02-13T22-47-19.235Z.log](https://github.com/user-attachments/files/18790992/meshtastic-log-2025-02-13T22-47-19.235Z.log)
> 
> Thanks for the logs. Could you capture them just after a reboot, too?

No problem but im pretty sure that its the connectors fault as it appers to be damaged a bit, where the connector makes conntact with the pins, ill resolder it and try again, possible that thats why it does not work

fifieldt, 2025-02-16:
Thanks for the update!

Sabering1, 2025-02-22:
It was the connector, closing

closed 2025-02-22 by Sabering1