meshtastic/firmware #5974: [Bug]: Charging Flag False when it should be true.

vlturner, 2025-02-01:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.5.20

### Description

I can see the unit charging on the web interface while usb cable is plugged in over time but the USB flag & charging flag is showing false. Also consider adding release to the JSON report Full Release ID is missing in the apps.

### Relevant log output

```Shell
{"data":{"airtime":{"channel_utilization":1.66833329200745,"periods_to_log":8,"rx_all_log":[0,0,0,0,0,0,0,0],"rx_log":[0,0,0,0,0,0,0,0],"seconds_per_period":3600,"seconds_since_boot":59846,"tx_log":[10522,12764,16228,12788,2742,10407,7551,12951],"utilization_tx":0.415388882160187},"device":{"reboot_counter":101},"memory":{"fs_free":1019904,"fs_total":1048576,"fs_used":28672,"heap_free":64864,"heap_total":261712,"psram_free":0,"psram_total":0},"power":{"battery_percent":49,"battery_voltage_mv":3717,"has_battery":"true","has_usb":"false","is_charging":"false"},"radio":{"frequency":906.875,"lora_channel":20},"wifi":{"ip":"192.168.0.11","rssi":-15}},"status":"ok"}
```

fifieldt, 2025-02-02:
Thanks for writing in. Can I confirm by "see the unit charging", you're seeing the charging icon in the web interface or through the app? However, the MQTT Json content shows the wrong values?

vlturner, 2025-02-03:
Yes the weird thing is now that it hit 100% it changed them, but during charging it doesn’t say connected or charging, but you see the battery charge growing.

 

{"data":{"airtime":{"channel_utilization":0,"periods_to_log":8,"rx_all_log":[0,0,0,0,0,0,0,0],"rx_log":[0,0,0,0,0,0,0,0],"seconds_per_period":3600,"seconds_since_boot":77276,"tx_log":[5911,11294,22209,19822,17571,12877,15298,18662],"utilization_tx":0.291888892650604},"device":{"reboot_counter":103},"memory":{"fs_free":1019904,"fs_total":1048576,"fs_used":28672,"heap_free":74892,"heap_total":264176,"psram_free":0,"psram_total":0},"power":{"battery_percent":100,"battery_voltage_mv":4296,"has_battery":"true","has_usb":"true","is_charging":"true"},"radio":{"frequency":906.875,"lora_channel":20},"wifi":{"ip":"192.168.0.11","rssi":-2}},"status":"ok"}

closed 2025-03-28 by github-actions[bot]