meshtastic/firmware #5979: [Board]: Adding TTGO LoRa32 V2.1 _ 1,6 TCXO Version?

Burny08, 2025-02-02:
### SOC

ESP32

### Lora IC

LILYGO 868/915Mhz Modell: LORA32 TCXO

### Product Link

https://de.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.62.7a875c5fXpc2yw&gatewayAdapt=glo2deu

### Description

I have been informed by the manufacturer that the TCXO variant is not supported.

The Modell is: T3 V1.6.1 with SX1276 TCXO

Unfortunately I am not a programmer myself, but he told me that this board can probably be added.
Here is the message from the manufacturer:

Hello, the meshtastic firmware is not adapted to the TXCO version yet.
You can add the below sentences in the source code to make it work.
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1276/SX1276_Receive_Interrupt/boards.h

Or to test hardware function, you can upload the below firmware to check :
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware/hardware_test/T3_V1.6.1_TXCO
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware#to-write-firmware-please-check-the-corresponding-link-below

[image]

Thanks!!

formtapez, 2025-02-03:
Thanks - i need that, tool.
In firmware-esp32-2.5.16.f81d3b0 there was a special "tcxo"-version for that board - unfortunately that is missing in newer versions.

formtapez, 2025-02-03:
Actually, this device-model is present in the actual firmware tree, but it is not included in the prebuild binaries.
When compiling yourself, you can select that TCXO-Firmware and it works.

Burny08, 2025-02-03:
> Actually, this device-model is present in the actual firmware tree, but it is not included in the prebuild binaries. When compiling yourself, you can select that TCXO-Firmware and it works.

And that's exactly the problem. Unfortunately, I don't know how to compile it myself. Is there a way to get this model into the prebuild binaries?

vanish125, 2025-02-12:
@Burny08 

[tlora-v2-1-1_6-tcxo.tar.gz](https://github.com/user-attachments/files/18768321/tlora-v2-1-1_6-tcxo.tar.gz)

Burny08, 2025-02-15:
> [@Burny08](https://github.com/Burny08) I build 2.5.20.4c97351 [tlora-v2-1-1_6-tcxo.tar.gz](https://github.com/user-attachments/files/18768321/tlora-v2-1-1_6-tcxo.tar.gz)

Thank you! I can't get around to it at the moment, but I'll test it when I get the chance. 
What option is there to include the hardware as standard? 
I usually ask for a new build every 1-2 years :D

vanish125, 2025-03-11:
@Burny08 Standard build don't include unsupported hardware. You will need to build it yourself, or ask anywhere.

formtapez, 2025-03-11:
> [@Burny08](https://github.com/Burny08) Standard build don't include unsupported hardware. You will need to build it yourself, or ask anywhere.

That model is well supported. It is selectable in the device-tree and works great!
The build-script for the github-releases just has to be updated, so that the model is included.

closed 2025-05-04 by github-actions[bot]