meshtastic/firmware #5980: [Bug]:

brebtatv, 2025-02-02:
### Category

Hardware Compatibility, Other

### Hardware

Rak4631

### Firmware Version

2.5.20.4c97351

### Description

The LTR390V sensor is not identified. Protobuf supports it, but I2C does not identify it.
Its address is found at 0x53, see the log excerpt.

### Relevant log output

```Shell
INFO  | ??:??:?? 1 Scan for i2c devices
DEBUG | ??:??:?? 1 Scan for I2C devices on port 1
INFO  | ??:??:?? 1 MAX17048 found at address 0x36
INFO  | ??:??:?? 1 SSD1306 found at address 0x3c
INFO  | ??:??:?? 1 SSD1306 found at address 0x3c
DEBUG | ??:??:?? 1 0x4 subtype probed in 2 tries 
INFO  | ??:??:?? 1 Device found at address 0x53 was not able to be enumerated
INFO  | ??:??:?? 1 BME680 found at address 0x77
INFO  | ??:??:?? 1 3 I2C devices found
```

gjelsoe, 2025-02-03:
Seems there are made some preparation for this sensor but not finished yet.

fifieldt, 2025-02-08:
Just confirming @gjelsoe 's conclusion - while the sensor was added to protobufs, there is not any support for it in code at the moment.

I've added a patch that will make it detected: https://github.com/meshtastic/firmware/pull/6009

but it won't do anything until the full sensor support is added.

brebtatv, 2025-04-03:
Please add support, half a year complete weather station. Thank you

dmarman, 2025-05-22:
It should be stated in the docs that the LTR390 is still in development and not supported yet. I just bought one to figure out it's not implemented.

https://meshtastic.org/docs/configuration/module/telemetry/

dmarman, 2025-05-29:
@thebentern should we open an issue for the Android (probably iOS too) client to display the UV index card within environment metrics? Right now the Android client is displaying an empty space for the UV card, so for the end user this is still broken.

PR #6872 (2025-05-29): Added full support for LTR390UV readings of UV and Lux
This PR fixes #5980 and gives full support at the firmware level for reading UV and Lux values with the sensor LTR390UV. 
The hardware sensor has been tested and the values compared with online weather stations in sunny Barcelona.

The different UI clients must update to show the UV reading. Currently with Android app, UV values will not show up however Lux readings do appear.

Due to the nature of the sensor. Lux and UV readings must be done in two passes. So for the clients, UV and Lux values are updated every two minutes.

## 🤝 Attestat [... cut]