meshtastic/firmware #5982: initNonce extraNonce position

dizcza, 2025-02-03:
Shouldn't this line

https://github.com/meshtastic/firmware/blob/3a34f8beaff09f494e3996f059737d10d3af75f0/src/mesh/CryptoEngine.cpp#L258

be

```c
memcpy(nonce + sizeof(uint64_t) + sizeof(uint32_t), &extraNonce, sizeof(uint32_t));
```
?

fifieldt, 2025-02-04:
@jp-bennett might know

esev, 2025-02-04:
See also:
- https://github.com/meshtastic/firmware/issues/5543
- https://github.com/meshtastic/firmware/pull/5613

jp-bennett, 2025-02-04:
I definitely need to do some better documentation of this. The problem is that it is a mess, and we've had to work to keep it backwards compatible. The Packet ID is actually only 32 bits. I think the draft patch from @esev gets it right. I didn't have time to dive deeply into that when he posted it, and honestly forgot to go back and look again.

dizcza, 2025-02-04:
I see. The draft PR clarified my question.

Well, it's been a month since it was created.

closed 2025-02-04 by dizcza