meshtastic/firmware #5984: [Bug]:

rexido1, 2025-02-03:
### Category

Hardware Compatibility

### Hardware

Linux Native

### Firmware Version

2.5.21.3a34f8be

### Description

On the old version (v2.5.19.9afadde2) of the firmware it worked (but it just wrote the message Warning, cannot claim pin GPIO233St16invalid_argument) and on the new version it writes an error and does not start. Orange Pi Zero 2W board ([pinout](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_2W#40pin_SPI_test_method))
pin233 - SPI1_CS1

### Relevant log output

```Shell
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
MAC ADDRESS: 24:AB:CC:FD:98:FB
Warning, cannot claim pin GPIO233St16invalid_argument
Error setting pin number 233. It may not exist, or may already be in use.
```

markbirss, 2025-02-03:
@rexido1 try just  commenting out cs

rexido1, 2025-02-03:
Old firmware work 

> [@rexido1](https://github.com/rexido1) try just without the cs pins defined in yaml

Meshtastic started 
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
MAC ADDRESS: 24:AB:CC:FD:98:FB
DEBUG | ??:??:?? 0 Upgrade time to quality NTP
DEBUG | 17:34:39 0 Read RTC time as 1738596879
INFO  | 17:34:39 0

//\ E S H T /\ S T / C

DEBUG | 17:34:39 0 Filesystem files:
DEBUG | 17:34:39 0  /. (0 Bytes)
DEBUG | 17:34:39 0  /.. (0 Bytes)
DEBUG | 17:34:39 0  /prefs/module.proto (125 Bytes)
DEBUG | 17:34:39 0  /prefs/channels.proto (278 Bytes)
DEBUG | 17:34:39 0  /prefs/. (0 Bytes)
DEBUG | 17:34:39 0  /prefs/.. (0 Bytes)
DEBUG | 17:34:39 0  /prefs/db.proto (21440 Bytes)
DEBUG | 17:34:39 0  /prefs/config.proto (288 Bytes)
INFO  | 17:34:39 0 Use /dev/i2c-3 as I2C device
INFO  | 17:34:39 0 Scan for i2c devices
DEBUG | 17:34:39 0 Scan for I2C devices on port 1

And why did it work without chip select?

closed 2025-02-03 by rexido1