meshtastic/firmware #5991: [Bug]: wio-e5 radio fails to initialize

dchokola, 2025-02-05:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.5.20

### Description

The SK1262 inside the STM32WLE5JC needs its TCXO initialized. Documentation says to set it 200 mV below Vcc.

This one-liner does the job:
```
--- a/src/mesh/STM32WLE5JCInterface.cpp
+++ b/src/mesh/STM32WLE5JCInterface.cpp
@@ -18,6 +18,8 @@ bool STM32WLE5JCInterface::init()
 {
     RadioLibInterface::init();
 
+    SX126xInterface::setTCXOVoltage(3.0f);
+
     lora.setRfSwitchTable(rfswitch_pins, rfswitch_table);
 
     if (power > STM32WLx_MAX_POWER) // This chip has lower power limits than some
```

### Relevant log output

```Shell

```

lyusupov, 2025-02-06:
@dchokola 

FYI: https://github.com/jgromes/RadioLib/issues/1174#issuecomment-2273792460

[image]

GUVWAF, 2025-02-15:
@dchokola Thanks for the report. According to [Seeed Studio's driver](https://github.com/Seeed-Studio/LoRaWan-E5-Node/blob/main/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio_driver.c), it should be set to 1.7V. This was set correctly before, but a newly introduced variable in the parent class caused it to be overridden. Should be fixed with #6063.

PR #6063 (2025-02-15): Fix STM32WL TCXO setting; enable logs and modules
Fixes #5991.

TCXO voltage had been set to 1.7V according to the [Seeed's driver](https://github.com/Seeed-Studio/LoRaWan-E5-Node/blob/main/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio_driver.c), but the `tcxoVoltage` class member introduced in #5948 got used instead from then on.
closed 2025-02-15 by thebentern