meshtastic/firmware #6010: 🚀 [Feature Request]: Scan QR code to enter private/public/admin key values

keith721, 2025-02-08:
### OS

iOS

### Description

Trying to manually enter 32-char/256-bit public/private/admin keys on the iPhone is quite a difficult task.  If the client.meshtastic.org web client could display a QR code containing the key values and the iPhone app could scan this code, it would make the task much easier.  Thank you for reading this request.

Alternately, could the app please change the key font color to something more easily readable?  Gray text on a white background is very low contrast and difficult to read.

### Participation

- [x] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Additional comments

I'm willing to contribute funds for this effort.  Probably not 100% of the funding, but at least 20%, if not more.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

thebentern, 2025-02-08:
What are the use cases here?
Is the use case backing up your own device's public / private key pairs for one?
Loading some shared public keys into a security config for remote admin on a router node via QR makes sense to me. Sort of like how we share symetrical keys on chanel, you could expose a QR on the client node, and scan it on the remote node?

Stephan-2025, 2025-03-06:
I also would found that feature very helpful, because of deploying a lot of similar nodes ... that would be much faster with that possibility (3 Public-Admin-Keys)