meshtastic/firmware #6014: [Bug]: battery is not reported on Repeater role.

Ccz119, 2025-02-08:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.5.20

### Description

On the latest version when set to role REPEATER  it no longer reports battery status via admin or any other method,even if requested. So when is on hilltop i have no idea how the battery is going. Also the 'charge'plug indicator is not reported also and it misleads that it it not charging. In other modes these issues are ok.

### Relevant log output

```Shell

```

fifieldt, 2025-02-08:
Hi, and thanks for using meshtastic :) 

This is expected behaviour for this role. There's a write up here on the various role types:
https://meshtastic.org/blog/choosing-the-right-device-role/

closed 2025-02-08 by fifieldt

Ccz119, 2025-02-09:
Thank you for the reply.