meshtastic/firmware #6017: [Bug]: T-deck does not display messages

bmw2002e3, 2025-02-09:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.5.18.89ebafc

### Description

I have several T-Deck Devices Primary channel is private channel, next channel is LongFast

When a message is sent to broadcast at either channel, the devices make an alert noise but do not display the message.

### Relevant log output

```Shell

```

fifieldt, 2025-02-09:
Hi @bmw2002e3 , thanks for sending in the report. 

We've got some exciting changes for the t-deck that we're testing right now -- would you like to give the new version a go? As part of the process unfortunately all your settings would be lost, and there could be new bugs, but it should provide a much better experience.

nktknzv, 2025-02-10:
Hi @fifieldt , can i try the new version? :)

bmw2002e3, 2025-02-10:
> Hi [@bmw2002e3](https://github.com/bmw2002e3) , thanks for sending in the report.
>
> We've got some exciting changes for the t-deck that we're testing right now -- would you like to give the new [...]

Absolutely! They are test devices for me anyway.

nktknzv, 2025-02-12:
The Bluetooth issue is no longer observed in version 2.5.22.d1fa27d Alpha.

closed 2025-02-12 by thebentern

fifieldt, 2025-02-27:
There is a new firmware preview for T-Deck that is a significant improvement, if you have this issue I recommend trying it out: https://meshtastic.org/blog/meshtastic-2-6-preview/ !