meshtastic/firmware #6039: [Feature Request]: ability to ignore Client Hidden

CraigwBowers, 2025-02-12:
### Platform

Cross-Platform

### Description

Currently it would appear the privacy of Client Hidden prevents these nodes from being contacts, allowing direct messages, or populating the node list. 

For me an annoying artifact seems to be that there is then no mechanism to ignore messages on LongFast primary channel from Client Hidden device mode nodes. 

I would prefer to see partial node list entries and the ability to perform normal node list actions on these such as filtering, ignoring. A bonus would be traceroute and direct messaging.

But minimum viable would be the ability to ignore annoying message noise from hidden nodes.

garthvh, 2025-02-12:
I don't think there is anything preventing the blocking of these nodes once they have sent a message.

CraigwBowers, 2025-02-13:
Perhaps I need to re-open as a bug then.
Have you tried to duplicate it?
The problem is that Client Hidden does not show up on the Node list.  No matter what.
Reproduce steps:

- Have 2 nodes in a private channel
- Set one node to Client hidden, and delete the node from the node list of the second device.
- You can send a message in the private channel from the client hidden node, exchange locations… send a message in primary channel LongFast…. None of that will populate the client_hidden in the other devices node list.  Since it’s not there, you cannot ignore it.
- When I filter the node list to role client_hidden, not one node shows on the list.  But I know there are Client_hidden around me, as close as 0 hops.  As they’re sending test messages in the primary LongFast channel, and never replying to anyone, or showing up in the node list.  Not showing up on the mesh map I understand if they aren’t sharing location.  But it seems like a bug for them to not show up in the node list.

fifieldt, 2025-02-13:
@CraigwBowers - thanks for your reply. I reopened the issue, and updated the title.  Is this android or IoS?

CraigwBowers, 2025-02-13:
IOS client 2.5.18 on iOS 18.3.1
RAK4361 and Seeed T1000-E both on 2.5.20

I can confirm the web client does see the client_hidden node.  It is only the iOS client that does not.

fifieldt, 2025-02-13:
@garthvh , wouldn't be surprised if there was some logic preventing client_hidden nodes appearing on the IoS app, could you take a look when you get a moment?

garthvh, 2025-02-13:
[image]

closed 2025-02-13 by garthvh