meshtastic/firmware #6050: 🐞 [Bug]: Node Name Updates Don't Reflect in App without Restarting

Xaositek, 2025-02-13:
### Firmware Version

2.5.18 (1355)

### What did you do?

I changed the "Owner" name is a node and noticed even after multiple hours the application doesn't reflect the update in my node list.

Old Name:
[image]

New Name:
[image]

### Expected Behavior

As NodeInfo updates are received by the radio, name updates are immediately reflected in the app

### Current Behavior

As NodeInfo updates are received by the radio, name updates are only updated if you force close the app and restart it.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

madeofstown, 2025-02-20:
> [@Xaositek](https://github.com/Xaositek) what happens if you disconnect / reconnect BLE? I'm wondering if I need to send the owner info back up over the PhoneAPI

With firmware v2.5.20 (and I'm sure earlier as well) nodes automatically reboot after a name change which would trigger a Bluetooth disconnect/reconnect. Did this change in a more recent firmware? What firmware are you using OP?

madeofstown, 2025-02-20:
Also, there are known issues with v2.5.18 and the `router_late` role

madeofstown, 2025-02-20:
> > > [@Xaositek](https://github.com/Xaositek) what happens if you disconnect / reconnect BLE? I'm wondering if I need to send the owner info back up over the PhoneAPI
> > 
> > 
> > With firmware v2.5.20 (and I'm sure earlier as well) nodes automatically reboot after a name change which would trigger a Bluetooth disconnect/reconnect. Did this change in a more recent firmware?
> 
> This is not about the directly connected node, this is about nodes in the node list

A change of a nodes name should still trigger a reboot whether it is connected to your phone or remotely administered or administered by somebody else... When a node boots up it should send out it's node information. I guess I'm missing something here...

Xaositek, 2025-03-13:
[image]
[image]
[image]
[Meshtastic Application Logs.csv](https://github.com/user-attachments/files/19237411/Meshtastic.Application.Logs.csv)
[name_change.pdf](https://github.com/user-attachments/files/19237419/name_change.pdf)

Xaositek, 2025-03-13:
@garthvh If you look at the attached screenshots you will notice.

Screenshot 1 (System Time: 4:52pm, Last Node Info: 4:48pm) - Shown Device Name is "Meshenger 4e91" and Actual Device Name is "Meshenger 4e91"

Screenshot 2 (System Time: 4:54pm, Last Node Info: 4:53pm) - Shown Device Name is "Meshenger 4e91" and Actual Device Name is "Meshenger 001"

Screenshot 2 (System Time: 4:54pm, Last Node Info: 4:54pm) - AFTER BLE DISCONNECT/RECONNECT - Shown Device Name is "Meshenger 001" and Actual Device Name is "Meshenger 001"

Logs are present showing node info received at 4:53pm:
DEBUG | 16:53:38 244338 [Router] handleReceived(REMOTE) (id=0xcd12d362 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x0 Portnum=4 rxtime=1741902818 rxSNR=7 rxRSSI=-20 hopStart=7 relay=
DEBUG | 16:53:38 244338 [Router] Module 'nodeinfo' wantsPacket=1
INFO  | 16:53:38 244338 [Router] Received nodeinfo from=0x2f224e91, id=0xcd12d362, portnum=4, payloadlen=76

Xaositek, 2025-03-13:
https://github.com/user-attachments/assets/fa511b70-9264-42e3-a82e-0d814f4e0c28
Adding a video also - I reverted the name back to showcase it. The name was changed around 5pm local time, here's it at 6pm. BLE disconnect/reconnect updated the name.

garthvh, 2026-09-03:
I can't find a path that makes this possible anymore, please retest

closed 2026-09-03 by garthvh

garthvh, 2026-09-03:
This was an Apple-client bug, not a firmware one, and it is fixed.

The firmware was forwarding the packet correctly. Over the mesh,
`PORTNUM_NODEINFO_APP` carries a bare `User` protobuf — `NodeInfoModule` is
declared as `ProtobufModule<meshtastic_User>` — but the iOS app only wrote the
name inside a branch that first decoded the payload as `NodeInfo`. That decode
throws on a `User` payload, so for a node the app already knew, the new name
was never applied. Newly discovered nodes were fine because that path decoded
as `User`, and a restart or reconnect fixed it because the NodeDB dump arrives
as real `NodeInfo` messages down a different path.

Fixed in meshtastic/Meshtastic-Apple#1648, first released in iOS/macOS
v2.7.13. Covered by a regression test using the node and names from this
report.

@Xaositek thanks for the log attachment — the iOS side of it is what showed
the packet was arriving and being handled, which is what located this. If you
still see it on a current app build, please say so and we will reopen on the
Apple repo.