meshtastic/firmware #6056: [Bug]: MQTT with TLS causes a reboot loop

esev, 2025-02-14:
### Category

WiFi

### Hardware

Heltec V3

### Firmware Version

v2.5.22.d1fa27d

### Description

I'm seeing a consistent crash right after WiFi startup when [TLS is enabled](https://meshtastic.org/docs/configuration/module/mqtt/#tls-enabled) for the MQTT server.

This was also reported here as well: https://www.reddit.com/r/meshtastic/comments/1iirozz/update_heltec_v3s_mqtt_ssl_and_crashing/ & https://www.reddit.com/r/meshtastic/comments/1ifpzb8/mqtt_with_tls_timeouts_no_setting/

I've observed that if I allow the node to continually crash for a while (>30 minutes) then eventually it will establish a TLS connection.

I believe the crash is happening due to the TLS connection requiring more stack space than a non-TLS connection. When WiFi connects, it triggers an immediate MQTT connection:

https://github.com/meshtastic/firmware/blob/9b46cb4ef08688a2f424c76d8425561e4f5db844/src/mesh/wifi/WiFiAPClient.cpp#L117-L118

This occurs within a `WiFiEvent` callback from the Espressif framework on the `arduino_events` task. The `arduino_events` task has only 4096 bytes of stack space. https://github.com/espressif/arduino-esp32/blob/9c0d59f9c4831dd03673f3a6294d21e6b13adb9b/libraries/WiFi/src/WiFiGeneric.cpp#L591

If I allow the node to crash for a while, at some random point this initial MQTT connection will fail. When that happens the MQTT retry logic begins. This retry logic happens from the main loop inside MQTT::runOnce.

https://github.com/meshtastic/firmware/blob/9b46cb4ef08688a2f424c76d8425561e4f5db844/src/mqtt/MQTT.cpp#L548

The main loop has a stack size of 8192 bytes https://github.com/espressif/arduino-esp32/blob/a7907cd07e605e4e9e37f0ec862e7da7a145fa38/cores/esp32/main.cpp#L17

**Possible fix**

MQTT::runOnce runs every 5 seconds when there is no WiFi connection.

https://github.com/meshtastic/firmware/blob/9b46cb4ef08688a2f424c76d8425561e4f5db844/src/mqtt/MQTT.cpp#L544-L546

I suggest removing the call to `mqtt->reconnect()` from WiFiAPClient.cpp (and ethClient.cpp), and allow `MQTT::runOnce` to reestablish the connection. I've tested with the following, and it is working for me.

```patch
diff --git a/src/mesh/wifi/WiFiAPClient.cpp b/src/mesh/wifi/WiFiAPClient.cpp
index 41de89794..d4a5dbf94 100644
--- a/src/mesh/wifi/WiFiAPClient.cpp
+++ b/src/mesh/wifi/WiFiAPClient.cpp
@@ -7,9 +7,6 @@
 
 #include "main.h"
 #include "mesh/api/WiFiServerAPI.h"
-#if !MESHTASTIC_EXCLUDE_MQTT
-#include "mqtt/MQTT.h"
-#endif
 #include "target_specific.h"
 #include <WiFi.h>
 #include <WiFiUdp.h>
@@ -111,12 +108,6 @@ static void onNetworkConnected()
 #endif
         APStartupComplete = true;
     }
-
-    // FIXME this is kinda yucky, instead we should just have an observable for 'wifireconnected'
-#ifndef MESHTASTIC_EXCLUDE_MQTT
-    if (mqtt)
-        mqtt->reconnect();
-#endif
 }
 
 static int32_t reconnectWiFi()
```

### Relevant log output

```Shell
[  8499][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (arduino_events)
Core  1 register dump:
PC      : 0x40384773  PS      : 0x00060036  A0      : 0x80382ac8  A1      : 0x3fcba5b0
A2      : 0x3fc9bca0  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x00000000  A11     : 0x3fc9bbe0  A12     : 0x7dc40000  A13     : 0x21c00000
A14     : 0x02c9bca0  A15     : 0x00ffffff  SAR     : 0x0000000e  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x420cc74a  LEND    : 0x420cc772  LCOUNT  : 0x00000000

Backtrace: 0x40384770:0x3fcba5b0 0x40382ac5:0x3fcba5f0 0x40381070:0x3fcba620 0x40381066:0xa5a5a5a5 |<-CORRUPTED
```

fifieldt, 2025-02-14:
Nice find, looks solid.

garthvh, 2025-02-15:
The public server also does not support tls, so we should also disable tls when users are using the public server

esev, 2025-02-15:
> The public server also does not support tls, so we should also disable tls when users are using the public server

Would it make sense for the admin command to fail if TLS is enabled and the default server is configured? I'm thinking the app might be able to supply some better context to the user in this case.

todd2982, 2025-02-15:
> The public server also does not support tls, so we should also disable tls when users are using the public server

I feel like this could be a separate PR, and more at the client app instead of firmware.

closed 2025-02-15 by thebentern

garthvh, 2025-02-15:
> > The public server also does not support tls, so we should also disable tls when users are using the public server
> 
> Would it make sense for the admin command to fail if TLS is enabled and the default server is configured? I'm thinking the app might be able to supply some better context to the user in this case.
> 
> Edit: Or maybe we _fix_ the user input. That way the next time the user views the settings within the app, they see the values that are actually in use.
> 
> Untested example:
> 
> diff --git a/src/modules/AdminModule.cpp b/src/modules/AdminModule.cpp
> index 7906b410b..151effeb0 100644
> --- a/src/modules/AdminModule.cpp
> +++ b/src/modules/AdminModule.cpp
> @@ -657,6 +657,22 @@ void AdminModule::handleSetModuleConfig(const meshtastic_ModuleConfig &c)
>          LOG_INFO("Set module config: MQTT");
>          moduleConfig.has_mqtt = true;
>          moduleConfig.mqtt = c.payload_variant.mqtt;
> +        if (moduleConfig.mqtt.address[0] == '\0' ||
> +            memcmp(moduleConfig.mqtt.address, default_mqtt_address, strlen(default_mqtt_address)) == 0) {
> +            if (moduleConfig.mqtt.tls_enabled) {
> +                LOG_WARN("Received MQTT config: tls_enabled is not valid for the default server. Setting to false.");
> +                moduleConfig.mqtt.tls_enabled = false;
> +            }
> +            String server = moduleConfig.mqtt.address;
> +            const int delimIndex = server.indexOf(':');
> +            if (delimIndex > 0) {
> +                const long parsedPort = server.substring(delimIndex + 1, server.length()).toInt();
> +                if (parsedPort != 1883) {
> +                    LOG_WARN("Received MQTT config: Invalid port '%d' used with the default server. Removing port.", parsedPort);
> +                    memcpy(moduleConfig.mqtt.address, default_mqtt_address, sizeof(default_mqtt_address));
> +                }
> +            }
> +        }
>          break;
>      case meshtastic_ModuleConfig_serial_tag:
>          LOG_INFO("Set module config: Serial");

The UI on iOS does not display the option, but none of the other clients do the validation so I think it needs to be an item that fails to save in the firmware with a client notification.

esev, 2025-02-16:
Will do in #6066, @garthvh

PR #6057 (2025-02-15): Fix MQTT over TLS crash loop
This PR removes the logic that establishes a MQTT connection immediately after WiFi is connected. The MQTT connection is established instead within 5 seconds of WiFi being connected by the `MQTT::runOnce` callback.

This is being done because the stack size of the task that notifies of WiFi connection state changes is too small to support TLS connections. See #6056 for further details.

Fixes #6056