meshtastic/firmware #6077: [Bug]: Non-TCXO LoRa module not detected by firmware newer than 2.5.18

tropho23, 2025-02-17:
### Category

Hardware Compatibility

### Hardware

DIY

### Firmware Version

>2.5.18

### Description

Recent changes made to fix TCXO radio detection for SenseCAP Indicator (https://github.com/meshtastic/firmware/pull/5948) broke non-TCXO radio detection for at least the Ra-01SH LoRa module.  I have installed this module on a so-called 'Faketec' using a Tenstar nrf52 ProMicro board.

**What doesn't work:**
Flashing _nrf52_promicro_diy_tcxo_ firmware >2.5.18 results in no LoRa radio detected in log output, and 'Critical Fault #3' displayed on device screen.

**What works:**
Flashing firmware 2.5.18, prior to the above mentioned PR 5948 change results in successful Ra-01SH detection.

**What also works, but is not ideal:**
Manually compiling the 'xtal' firmware variant (_nrf52_promicro_diy_xtal_) correctly identifies the Ra-01SH module for firmware >2.5.18 (2.5.22 daily build tested successfully today).

I recognize the intent to have only one firmware variant for _nrf_promicro_diy_ and detect the LoRa radio, which firmware 2.5.18 does successfully.  I recommend examining whatever changes were made to resolve SenseCAP Indicator TCXO detection in firmware 2.5.20 to see what might cause this issue.

### Relevant log output

```Shell

```

NomDeTom, 2025-02-17:
I can't reproduce this, unfortunately.

I've tried:

- 2.5.18
- 2.5.21
- 2.5.22

and in all cases I've managed to get `INFO  | ??:??:?? 4 SX1262 init success, XTAL, Vref 0.0V` from a pro-micro with RA-01SH.

tropho23, 2025-02-17:
I too have a ProMicro nrf52840 board (Tenstar Robot ProMicro clone?) and Ra-01SH 'AI Thinker' LoRa module.  According to documentation and other users this clone board is a 1:1 copy of the original ProMicro board.

I can reliably reproduce this issue as it gave me trouble for hours until I learned about the XTAL variant, which when flashed solves the problem.

I have assembled three DIY boards with this combination of board and LoRa module and all three encounter the radio detection failure issue with firmware >2.5.18, and all three detect the radio either with firmware 2.5.18 or >2.5.28 using manually-compiled firmware based on the _nrf52_promicro_diy_xtal_ variant.

I have ruled out soldering failure since all three boards suffer the same issue, and also I know my soldering skills are not bad :)

thebentern, 2025-02-17:
I would question the schematic of the faketec boards matching the variant potentially over the pro-micro. Can you link the Faketec boards you have?

tropho23, 2025-02-17:
> I would question the schematic of the faketec boards matching the variant potentially over the pro-micro. Can you link the Faketec boards you have?

This link, the newer v4 board: https://github.com/gargomoma/fakeTec_pcb/tree/main/gerbers

tropho23, 2025-02-17:
@NomDeTom helped me identify the issue: the black ProMicro boards must be an older version that does not power on or switch VCC as needed, while the red 'Tenstar Robot ProMicro' boards do.  I see 3v on VCC on the red boards, and 0v on these black ones.  Lucky for me I only have three of the black ones and sixty of the red ones :)

I recommend closing this issue as it is specific to an nrf52 board that people should not use when assembling Faketec boards.  There is no easy way to tell whether a board does or does not support VCC switching/activation besides buying one and testing it, unless the seller specifies that feature is supported.

Thanks all!

closed 2025-02-17 by thebentern