meshtastic/firmware #6083: [Bug]: Xiao Esp32 + Ina219 battery level override not working

nis0521, 2025-02-18:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

Latest Version

### Description

Hi guys,

Having problems having the battery level recognized in my Xiao Esp32 when I add on a INA219. 

I’ve made sure that all is wired correctly and the channel is set correctly. It does show up as Channel 3, would love to get the voltage recognized as battery as intended. :) 

### Relevant log output

```Shell

```

fifieldt, 2025-02-18:
Thanks for using meshtastic! To help us get to the source of the issue, would you be able to plug your device in with USB and go to https://flasher.meshtastic.org click open serial monitor. Then, reboot your device, let it run for 5-10mins and send the logs here!

caveman99, 2025-02-18:
this is tracked in https://github.com/meshtastic/firmware/pull/6070

nis0521, 2025-02-18:
Hey guys,

Do you still need that log?

syssi, 2025-02-19:
The log isn't needed anymore. #6070 will resolve the issue!

closed 2025-02-20 by fifieldt