meshtastic/firmware #6086: [Board]: Heltec SensorHub

Richard3366, 2025-02-18:
### SOC

ESP32

### Lora IC

SX1262

### Product Link

https://heltec.org/project/sensor-hub/

### Description

Hello, Meshtastic team.

SensorHub (HRI-3621) is an old project by the Heltec team, and the hardware portion of the device has now been open-sourced. We hope that the integration with Meshtastic will breathe new life into this project.

SensorHub is a device with ESP32S3+LoRa+sensor. Modular design allows for the replacement of sensor probes to adjust different projects. 
It comes with a built-in 1100mAh rechargeable battery and magnetic charging interface. The stainless-steel shell and high-strength plastic provide an IP65 protection level, allowing it to perform stably and excellently in complex environments.

[image]

[image]

caveman99, 2025-02-18:
Hello, this seems quite straightforward, judging from the schematics. About the sensor interface: Meshtastic natively supports only I2C connected sensors. If the sensor boards for this SensorHub use different technology and protocols, we can accomodate this in a custom firmware build, but support would be limited to this device because of firmware size contraints. Can you give me an idea which sensor boards you like to see support for and which protocols they use?

Richard3366, 2025-02-18:
> Hello, this seems quite straightforward, judging from the schematics. About the sensor interface: Meshtastic natively supports only I2C connected sensors. If the sensor boards for this SensorHub use [...]

The device mainboard retains both standard I2C and SPI interfaces. All sensors that comply with the standard I2C and SPI interfaces are supported.
Currently, we only offer a few default sensors to choose from. We are working on developing more standard-interface sensors based on the sensor models supported by Meshtastic.

Richard3366, 2025-02-18:
> Hello, this seems quite straightforward, judging from the schematics. About the sensor interface: Meshtastic natively supports only I2C connected sensors. If the sensor boards for this SensorHub use [...]

After a round of rigorous selection, we have retained only the temperature and humidity sensor, the three-axis sensor, and the barometric pressure sensor. Sensors like methane and TVOC have been eliminated due to their high power consumption! We are currently working on developing more other sensors.

caveman99, 2025-02-19:
SPI could be used in a future expansion, but most I2C low-rate sensors should work out of the box. Would you be able to provide a sample to validate the porting effort?

Richard3366, 2025-02-20:
> SPI could be used in a future expansion, but most I2C low-rate sensors should work out of the box. Would you be able to provide a sample to validate the porting effort?

Of course, we now have three types of sensors: temperature&humidity, atmospheric pressure and three axes. How many samples and sensors should I send you?

fifieldt, 2025-03-21:
@Richard3366 , did you successfully get in touch with @caveman99 ? 

I'm available to help too if there's a need.

caveman99, 2025-03-21:
@fifieldt i received two samples last tuesday, fun is commencing

fifieldt, 2025-03-21:
Enjoy!

caveman99, 2025-03-21:
added support with https://github.com/meshtastic/firmware/commit/0951fdd49b67c00c3c075b4236b12af7fb04d507