meshtastic/firmware #6089: [Feature Request]: Add a client notification for imprecise locations values outside of the ranges used by iOS and the public MQTT Server

garthvh, 2025-02-18:
### Platform

Cross-Platform

### Description

The range used is 11-14 32 is full precision

ianmcorvidae, 2025-02-19:
What's the benefit here? I guess I can see it for under-11 because those are pretty much just uselessly large, but 15-19 or so at least are still useful. There's not really much difference between higher than that and fully precise but they don't seem like a problem either. In the communities I'm part of & monitor, people who care about MQTT aren't using the public server because of the restrictions anyway. Perhaps this should be limited to those who are directly using the public server & have uplink enabled on the default channel?

garthvh, 2025-02-19:
To prevent imprecise locations that are too detailed to obscure your location.

ianmcorvidae, 2025-02-19:
If the proposal is to disallow imprecise values outside of the 11-14 range then I definitely oppose this. Many folks, perhaps even most, do not align their privacy thresholds with Apple policy or CCPA/GDPR legislation. Precision levels above 14 but less than 32 see significant use in my area.

garthvh, 2025-02-20:
Why would anyone be using imprecise locations that are easily discoverable?

esev, 2025-02-20:
I use a fixed position _near_ my home for my "home base" node, and then set a precision of 16. I want folks to know what neighborhood/area it is in, and I want them to also know that it is not a precise location. I think I set this either from the web UI or from an admin command.

I want folks to know there is a fixed node nearby, without giving away my exact house location.

garthvh, 2025-02-20:
> I use a fixed position _near_ my home for my "home base" node, and then set a precision of 16. I want folks to know what neighborhood/area it is in, and I want them to also know that it is not a [...]

16 has given away your position, with the Android app your position can be easily determined.

esev, 2025-02-20:
I have a fixed position set. But the fixed position at a park near my home. I want folks to know the park is not where the node is located, but is nearby, and so I set a precision of 16.

garthvh, 2025-02-20:
> I have a fixed position set. But the fixed position is at a park near my home. I want folks to know the park is not where the node is located, but is nearby, and so I set a precision of 16. I'm not trying to hide the fixed position that I've set. I'm just trying to let folks know that it isn't precise.

This would be an edge case that is lost, while better protecting user privacy.  More than half of meshtastic users won't even see your node in this case as iOS and the public mqtt don't show this at all.

fifieldt, 2025-02-20:
I do the same as @esev. It's also the approach a local community member who makes the solar nodes for interested volunteers suggests to newcomers. Though in Taipei we typically use 17 or 18 since it's so dense.

I've also seen a fun hide-and-seek variant using Meshtastic discussed that intended to use those 'mid' precisions :)

esev, 2025-02-20:
This near-precise information is useful to our local net. It helps identify where packets are being received without giving up our fixed node locations.

My main use-case isn't an app. It's our community meshview instance which is fed data from our local MQTT instance. It lets me see where my packets are being received from. 

[image]

ianmcorvidae, 2025-02-20:
The linked discussion reveals locations at any corner of four regions, regardless of precision. you would have exactly the same problem if you were at the corner of a precision-14 zone, or 11, or most anything. The very high precision numbers increase the number of places this happens, and that's all. 

I also use precision 16, without need of a fixed position, which in my low-density urban area is still about a hundred houses. With a half-mile major street grid, precision 16 is pretty much identical to cross streets, since those give a value accurate to about a quarter mile, 1320 feet.

Like I said, different areas and people have different privacy thresholds. I'm not surprised to learn that higher density places use higher numbers than 16, either. In Taipei I bet a precision 17 area has more people & households than my precision 16 one does!

ianmcorvidae, 2025-02-20:
also, Android allows quite a bit more than 11-15, unless you bullied the maintainers into destroying more features in the barely released beta from a couple days ago:

[image]

thebentern, 2025-02-20:
Seems like this one isn't quite as cut and dry as previously assumed with the 15+ precision range. We should definitely take care of the < 11 range now though. Those do nothing but create eye-sores on a map.

garthvh, 2025-02-20:
> The linked discussion reveals locations at any corner of four regions, regardless of precision. you would have exactly the same problem if you were at the corner of a precision-14 zone, or 11, or [...]

With both 16 and 18 and positions from a device gps on an android phone my location is easily pinpointed once there are a couple hundred locations.

caveman99, 2025-02-20:
Can we agree on the precision setting being a case of informed cconsent with a CCPA/GDPR-friendly default? Nobody should be prevented from setting something they intend to set, but with a sensinble privacy-default. The latter is given, the former not. CCPA/GDPR is not a ball-and-chain approach, if someone absolutely wants to broadcast their position, they should be allowed to.

esev, 2025-02-20:
We can help users better understand the setting too. If we expect that 11-14 is considered an imprecise location and that 15-32 is considered a precise location, maybe that is something the UI can show. As the slider crosses between 14 & 15 the app could change the displayed text to show that the boundary between imprecise/precise has been crossed. It would help with making an informed decision.

Changing the defaults in the app and the firmware can help too. It still seems the Android app defaults to a precise location when the position is first enabled. I personally would have been more comfortable with a default of 11 after enabling position reporting. Then it can be changed to be more precise as desired.

ianmcorvidae, 2025-02-20:
Got away from the internet last night, will respond to several things at once here:

> Can we agree on the precision setting being a case of informed cconsent with a CCPA/GDPR-friendly default? Nobody should be prevented from setting something they intend to set, but with a sensinble [...]

Agreed, 100%.

> With both 16 and 18 and positions from a device gps on an android phone my location is easily pinpointed once there are a couple hundred locations.

I'd be curious to see the method of this for any case except sitting on an intersection of 4 precision zones or near an edge, which is the case in the linked discussion. If your position is far enough from the edge of any precision area that your GPS inaccuracy doesn't mean you appear to leave the zone, that shouldn't reveal information beyond the configured precision, and if you're near only one edge and not a corner it should only reveal that you're within GPS inaccuracy range of that one edge. If the zone is smaller than GPS inaccuracy, it'll probably show you in the zone you're actually in more often than inaccurate ones, and maybe you could statistically reason about which edges/corners someone is closer to based on how many appear in each of the 8 surrounding zones, but that's a far cry from it pinpointing the location. That case is more like "statistics could render the location more precise than your setting". Which does seem like something worth informing users about.

(And also to note again, when near edges and corners it doesn't matter what precision level you've set; sitting on the corner of 4 precision-11 zones will just create a bigger X over your basically-precise location in the log)

I think we could probably improve on this problem with some sort of flapping detection near edges of precision zones. But that would need some engineering, of course. Given that it applies to any precision setting, not just more precise settings, I think we probably should do that work. I'll see if I can put together a proposal.

> We can help users better understand the setting too. If we expect that 11-14 is considered an imprecise location and that 15-32 is considered a precise location, maybe that is something the UI can [...]

I like this idea a lot. I'd say that 20-32 are basically fully-precise, 15-19 or so are maybe "mostly precise", with 11-14 as truly imprecise. I think we should add the "mostly precise" category to clarify why someone might choose those rather than just setting it to full precision.

Possible 3.0 idea would be taking advantage of position precision to pack positions into less space over the air, somewhat relatedly. If the position was a single `bytes` field, full precision would take the same space as today (1 byte field tag, 1 byte length, 8 bytes actual data (4 per lat/lon), vs. today where it's two different field tags and two fixed32s that take 4 bytes each), but with position precision of 17-24 it could use 8 bytes (2+6), and with 9-16 could use only 6 (2+4). This would be another reason to use especially 16 or 24 instead of 32.

> Changing the defaults in the app and the firmware can help too. It still seems the Android app defaults to a precise location when the position is first enabled. I personally would have been more [...]

Also agreed. I think we set something like that for the default channel in firmware, but defaulting to something like that in the apps would improve this. As a default that can be overridden, it can also let us be more consistent between the default channel and new/private channels -- have everything default to whatever value we like but be configurable.

closed 2025-04-15 by github-actions[bot]

garthvh, 2025-05-21:
Looks like  12-15 or  13-15 will be the GDPR / CCPA values for imprecise location.

sandroshu, 2026-09-08:
Make this an OPTION not an enforcement. This is bad.

d3xt3r01, 2026-09-11:
how is this completed? :| it's clamped without an option to do otherwise ( as many if not all in my area do and here also )...
I don't really care about what apple thinks, and I'm fine with sharing the node location, I choose to!

willp618, 2026-09-18:
believe this is the issue im facing since updating to 2.8.0 where no long can set location precision less than 2,392ft or 0.7km this puts me well outside of my town 

personally i think it should not be forced it on people it should be a setting that already turned on so those that want to have more precision can have more precision by turning off a switch that gives lots of warning 

i would also note some areas like mine make heavy use of local_only decode which makes using a private channel hard 
and i bet my area is not the only area setup this way 

because of this i will be going back to 2.7.26 

running 2.8.0.47db0e3 on EU_868 Longfast on a wistag using android app 2.8.2(29322131) fdroid

sandroshu, 2026-09-18:
> because of this i will be going back to 2.7.26

For the time being we pathced this madness out from 2.8.1 (you can basically edit src/mesh/PositionPrecision.h and set MAX_POSITION_PRECISION_PUBLIC_KEY to be 32).

To avoid doing this I might just abandon this project and chose something else that actually does not try to make my life harder than it should be.. like meshcore or reticulum

willp618, 2026-09-18:
thank you for letting me know this

will have ago later on today