meshtastic/firmware #6093: [Bug?]: cannot find -l algobsec

luisfromero, 2025-02-19:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.5.22

### Description

First of all, sorry for my lack of experience. This is my first time contributing to GitHub.

I just wanted to mention that, after carefully following the instructions at [Meshtastic Firmware Build](https://meshtastic.org/docs/development/firmware/build/), I ran into the same issue on two different computers (both running Windows 11) and with two different variants (LoRa v2.1 and TBeam-S3-Core):
```swift
c:/users/felipe/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lalgobsec
```
I was able to fix it easily by adding this line to the corresponding platformio.ini:
```makefile
build_flags = 
  -L"D:/onedrive/proyectos/meshtastic/firmware/.pio/libdeps/tbeam-s3-core/bsec2/src/esp32s3"
```
I don’t know the root cause of the issue since my knowledge doesn’t go that far, but I saw that someone else had the same problem—I think they mentioned it on Reddit—but they didn’t get a response.

Thanks, and sorry if this is out of place.

### Relevant log output

```Shell

```

fifieldt, 2025-02-19:
I don't immediately know the answer, but just wanted to say welcome to github (you did it right!) and thank you for using Meshtastic.

caveman99, 2025-02-19:
The build flags are added by the processor target automatically and should work on windows. there's something wrong with your setup. For nRF52 we needed to add this linker flag manually since the floating point capability was not correctly detected.

`-L "${platformio.libdeps_dir}/${this.__env__}/bsec2/src/cortex-m4/fpv4-sp-d16-hard"`

You can use something similar to work around your issue, but please don't submit this as a patch since it's working correctly in general.

closed 2025-02-19 by caveman99