meshtastic/firmware #6097: [Feature Request]: click sequence send canned message.

Alex-Pennington, 2025-02-19:
### Platform

ESP32

### Description

SOS/MAYDAY feature.
Use case: stand alone tracker, i.e. no phone connected.  Search and Rescue personnel carry a tracker on person and use it to request assistance.
Proposed method: A module that when enabled watches for a triple click or other sequence and send a canned message i.e. 'SOS'

thebentern, 2025-02-21:
https://meshtastic.org/docs/configuration/module/canned-message/#input-source

closed 2025-02-21 by thebentern

Alex-Pennington, 2025-02-22:
Thanks for your reply I will look into it.

Alex-Pennington, 2025-02-22:
My request was to use an already existing hardware button, i.e. T-Beam with three buttons, and map a click sequence as an input source for the canned message module and the detection module.  Similar this:

https://github.com/meshtastic/firmware/blob/fe1ced748046f9a2f0c4e544177ac15bd51f609f/src/ButtonThread.cpp#L194C1-L206C23