meshtastic/firmware #6102: [Bug]: config.d yaml overwrites all key's values in config.yaml

pdxlocations, 2025-02-19:
### Category

Other

### Hardware

Linux Native

### Firmware Version

2.5.21

### Description

Yaml keys in config categories may overwrite the entire category, while the intention is to merge them.
Discord discussion here: https://discord.com/channels/867578229534359593/972586878337433670/1341855284728958977

with `/etc/meshtasticd/config.yaml` including:
```
General:
  MaxNodes: 500
  MaxMessageQueue: 100
  ConfigDirectory: /etc/meshtasticd/config.d/
#  MACAddress: AA:BB:CC:DD:EE:FF
#  MACAddressSource: eth0
```

and `/etc/meshtasticd/config.d/test.yaml` including:
```
General:
  MaxMessageQueue: 100
```

The nodedb size does not honor settings from config.yaml and defaults to 200:

```
Feb 19 13:34:43 meshdev meshtasticd[73564]: INFO  | 21:34:41 4093 [Router] Node database full with 200 nodes and 4294967295 bytes free. Erasing oldest entry
Feb 19 13:34:43 meshdev meshtasticd[73564]: INFO  | 21:34:41 4093 [Router] Adding node to database with 200 nodes and 4294967295 bytes free!
```

delete test.yaml and the nodedb size may again exceed the default 200:

```
Feb 19 14:14:24 meshdev meshtasticd[74851]: INFO  | 22:14:03 1194 [Router] Adding node to database with 202 nodes and 4294967295 bytes free!
```

### Relevant log output

```Shell

```

Szewcson, 2026-09-26:
I'm also interested in that fix.

Szewcson, 2026-09-26:
One practical consequence of the current section-level override behavior is that it makes it difficult to customize USB device selectors such as "USB_VID", "USB_PID", or "USB_Serialnum" while still inheriting the rest of an upstream hardware profile.

For example, with a MeshToad/MeshTadpole profile I would like to keep the complete upstream hardware definition, but override only the USB selector:

Lora:
  USB_PID: 0x1234
  USB_Serialnum: "13374201"

The intention would be to inherit the rest of the stock "Lora:" section, including:

Module: sx1262
CS: 0
IRQ: 6
Reset: 2
Busy: 4
RXen: 1
DIO2_AS_RF_SWITCH: true
DIO3_TCXO_VOLTAGE: true
spidev: ch341
USB_VID: 0x1A86

With the current behavior, however, a later "Lora:" section is not a recursive field-level overlay. Values omitted from the override can be reset/defaulted instead of being inherited from the previously loaded profile.

This is especially problematic for "Module: auto".

A typical desired flow would be:

base config:
  Module: auto
  USB_VID: 0x1A86
  USB_PID: 0x1234
  USB_Serialnum: 13374201

        ↓

autodetection finds the customized CH341 device

        ↓

MESHTOAD product string selects
lora-usb-meshtoad-e22.yaml

        ↓

stock MeshToad hardware settings are applied,
while the user-specified VID/PID/serial remain overridden

Today that last step does not work as an overlay. The stock MeshToad profile contains its own "USB_PID: 0x5512" and does not define "USB_Serialnum", so loading the profile can replace/reset the values that were needed to find the customized USB device in the first place.

This means users cannot cleanly say:

«“Use the standard upstream hardware profile, but override these one or two fields.”»

Instead, they have to copy the complete upstream profile and modify it, which creates configuration drift and means future upstream changes to that hardware profile are not inherited automatically.

A recursive/deep merge of mappings would solve this nicely:

- mappings: merge recursively;
- scalars: later value wins;
- lists: replace as a whole, unless another explicit merge behavior is defined.

Then small overrides such as VID/PID/serial, GPIO changes, or other hardware-specific adjustments could be layered on top of upstream profiles without duplicating the entire profile.