meshtastic/firmware #6107: [Feature Request]: Client-Fast

MeshBamaMark, 2025-02-20:
### Platform

NRF52, ESP32

### Description

This request is based on the premise that the rebroadcast rate is as follows:

Client - Standard Rate
Router and Repeater - Fastest and the same rate

If this is the case, it would be nice to have a Client-Fast, that had a rebroadcast rate between the standard client and the Router/Repeater role.

Here is the thought process ...

Client used for non-mobile on top of houses and in trees
Client-Fast would be used in higher locations on towers and hills
Routers/Repeaters used only at the highest locations, and very few of them.

Thank you for your thoughts.

vidplace7, 2025-02-20:
Doesn't `ROUTER_LATE` accomplish this already?

GUVWAF, 2025-02-20:
Any `CLIENT` will be a fast client when they're likely to reach more nodes due to their location, since the delay is based on signal quality. 
Indeed, as vidplace7 mentions, `ROUTER_LATE` might be a good candidate for this. It makes sure it always rebroadcasts once, whereas `CLIENT`s won't if they hear another doing so already.