meshtastic/firmware #6120: [Feature Request]: External notification based on message

Patrick762, 2025-02-22:
### Platform

Cross-Platform

### Description

It would be great if you could assign incoming messages to external notifications.
So for example, if the node receives "relay on" it turns the assigned gpio pin on. This should be however only possible for an admin node to trigger.

thebentern, 2025-02-22:
We sort of have this with the ASCII bell based alerts in ext. notifications. I would recommend starting there.

GUVWAF, 2025-02-22:
There's also the [Remote Hardware module)(https://meshtastic.org/docs/configuration/module/remote-hardware/), although it's currently excluded from the firmware by default. It (for now) only work with the Python CLI.