meshtastic/firmware #6141: [Feature Request]: Admin blocking of settings options via app (iOS and Android) for public Nodes (iOS and Android)

MALAONE1, 2025-02-24:
### Platform

NRF52, other

### Description

Perhaps in the future it will be possible to build an option into the firmware to allow node owners to block all settings via app for publicly accessible nodes where multiple people use a node. The only thing that should be selectable is the channel specified by the node owner. This makes it possible to make a node available to multiple people without anything being changed in the settings.

b8b8, 2025-02-24:
Managed mode.

KamikazeYasha, 2025-02-24:
This is a desirable function in case a second admin is registered who, for example, should only restart the node.

closed 2025-02-24 by garthvh

b8b8, 2025-02-24:
> [@b8b8](https://github.com/b8b8) No, it's not the same. Please read again. I would like to give others public access to the node with their smartphone without being able to change settings. I'm not talking about admin access with another node

Managed Mode is NOT admin access. Have you tried it? It does exactly what you want, it locks out the settings. Also referred to as old people mode or kid mode.

MALAONE1, 2025-02-24:
@b8b8 In the last attempt, Bluetooth and Serial were disabled in managed and could only be accessed via Admin Node

MALAONE1, 2025-02-24:
@b8b8 I know what you mean, but you can still delete channels or ignore nodes, set markers on the map, etc. If that were still blocked, it wouldn't be a problem. We wanted to set up a public Meshtastic radio station so that people can test it without their own device.

garthvh, 2025-02-24:
Managed mode prevents using admin messages, waypoints are just packets and are not admin messages, existing functionality already exists for this use case.

MALAONE1, 2025-02-24:
@garthvh  . Okay, and how can you prevent channels from being deleted or nodes from being ignored when the node is used publicly? Is there a solution for this? I have seen that this still works despite the Managed Mode.

MALAONE1, 2025-02-24:
@garthvh Okay thanks for the information

KamikazeYasha, 2025-02-24:
Ei Ei.... Vorsischt die Hesse komme! 😂✌️

MALAONE1, 2025-02-24:
@KamikazeYasha  I like the Meshtastic system, so I want to spread it🙂✌️

KamikazeYasha, 2025-02-24:
Yes @MALAONE1 , we will use this feature to introduce Meshtastic to people. Maybe at an event.